pftf/RPi4 #118: Hardcoded UARTs in ACPI

worproject, 2020-12-09:
On the Pi 4, PL011 is hardcoded in DBG2 & SPCR, and mini UART is used for the Bluetooth device. 
This doesn't make sense when booting Windows, because GPIO 14 / 15 are set to ALT5 (mini UART) by default. So we end up with a debug port that's not exposed on the GPIO header.

I also want to use PL011 for Bluetooth instead as it's better than mini UART.

samerhaj, 2020-12-09:
This is already referenced as a working item here  https://github.com/pftf/RPi4/issues/80

pbatard, 2020-12-09:
I asked for a separate issue to be created to better track this item.

Basically, we have the following ACPI entries that are hardcoded according to the platform:
* [`Dbg2.aslc`](https://github.com/tianocore/edk2-platforms/blob/8f088a1224ede7f3d1dd97d79399eac3fc40af8b/Platform/RaspberryPi/AcpiTables/Dbg2.aslc#L26-L39)
* [`Spcr.aslc`](https://github.com/tianocore/edk2-platforms/blob/8f088a1224ede7f3d1dd97d79399eac3fc40af8b/Platform/RaspberryPi/AcpiTables/Spcr.aslc#L22-L30)
* [`Uart.asl`](https://github.com/tianocore/edk2-platforms/blob/8f088a1224ede7f3d1dd97d79399eac3fc40af8b/Platform/RaspberryPi/AcpiTables/Uart.asl#L129-L133)

Instead, these tables should be updated to reflect if the miniUART overlay has been applied by the user (similar to what we are doing [here](https://github.com/tianocore/edk2-platforms/blob/master/Platform/RaspberryPi/Library/DualSerialPortLib/DualSerialPortLib.c#L89-L95) to properly set either PL011 or miniUART to be used for serial/BT on the Pi 4.

pbatard, 2020-12-09:
Hopefully, we can do something along the lines of what Jeremy did in tianocore/edk2-platforms@e372412acf3d8d0147f57464c84f8031f41282af to set these ASL values at runtime. Otherwise, we may have to look into using [DynamicTablesPkg](https://github.com/tianocore/edk2/tree/master/DynamicTablesPkg)...

samerhaj, 2021-04-12:
A patch for this has already been sent to TianoCore EDK2, not merged yet: https://edk2.groups.io/g/devel/topic/81572763#73237

sunnywang-arm, 2021-06-14:
The changes got merged as the following commits, so close it 
- https://github.com/tianocore/edk2-platforms/commit/ea709f165df7899d0a9602fc9f65b824baeac490
- https://github.com/tianocore/edk2-platforms/commit/eb195ee197b8f56afebb7b33126ab0638f8c4451
- https://github.com/tianocore/edk2-platforms/commit/086a3a3ce6c42c859ee5943eb2c4b6edcefdc241
Thanks, @pbatard @ardbiesheuvel @jlinton @mariobalanica @samerhaj

closed 2021-06-14 by sunnywang-arm