pftf/RPi4 #119: How can I fix wlan/Bluetooth?

gesangtome, 2020-12-18:
Nice work, I successfully installed fedora through the ISO of aarch64. :+1: 
However, I cannot find the interface of brcm wlan0 through command 'ifconfig' or 'lshw -C network'. I think wireless and Bluetooth are useful in most cases, can I do something?

samerhaj, 2020-12-18:
You may find these comments useful: https://github.com/pftf/RPi4/issues/26#issuecomment-742040031

gesangtome, 2020-12-18:
@samerhaj Switch AICP to DT?

bzy-080408, 2020-12-26:
And when I run Ubuntu 20.04 and 20.10 on it, I also cannot use the wlan.

bzy-080408, 2020-12-26:
I think you can build the kernel by the kernel. org

gesangtome, 2020-12-26:
> I think you can build the kernel by the kernel. org

I think it has nothing to do with the kernel. The latest fedora already contains the firmware of BCM and the latest kernel. Even if I manually modprobe brcmfmac, the corresponding network card interface does not appear. Frustrated.

gesangtome, 2020-12-26:
> @samerhaj When I switch to DT, it seems that the wlan interface still does not appear.

jlinton, 2021-01-04:
For it to work with a linux/DT you also need to change the default sd pin muxing, which is in the current firmware but you really need the patches I posted to edk2 in order for that selection to work properly.

jlinton, 2021-01-13:
I've also just gotten patches merged to the latest linux-firmware repos which link the correct firmware based on the DMI strings we export.

Basically just hang on a couple weeks and the 5.12 should have support for the SD/Wifi in ACPI. 

Right now you need this set: https://lkml.org/lkml/2021/1/8/798

and an edk2 with the ACPI patches which hopefully get merged in the next week or so.

SquallATF, 2021-01-19:
> 
> 
> I've also just gotten patches merged to the latest linux-firmware repos which link the correct firmware based on the DMI strings we export.
> 
Maybe can modify  `drivers/net/wireless/broadcom/brcm80211/brcmfmac/dmi.c` to load the correct firmware.
I have tested, this modify can work.

```patch
--- a/drivers/net/wireless/broadcom/brcm80211/brcmfmac/dmi.c
+++ b/drivers/net/wireless/broadcom/brcm80211/brcmfmac/dmi.c
@@ -40,6 +40,10 @@ static const struct brcmf_dmi_data pov_t
 	BRCM_CC_43340_CHIP_ID, 2, "pov-tab-p1006w-data"
 };
 
+static const struct brcmf_dmi_data raspberrypi_4b_data = {
+	BRCM_CC_4345_CHIP_ID, 6, "raspberrypi,4-model-b"
+};
+
 static const struct dmi_system_id dmi_platform_data[] = {
 	{
 		/* ACEPC T8 Cherry Trail Z8350 mini PC */
@@ -115,6 +119,15 @@ static const struct dmi_system_id dmi_pl
 		},
 		.driver_data = (void *)&pov_tab_p1006w_data,
 	},
+	{
+		/* Raspberry Pi 4 Model B */
+		.matches = {
+			DMI_EXACT_MATCH(DMI_SYS_VENDOR, "Raspberry Pi Foundation"),
+			DMI_EXACT_MATCH(DMI_PRODUCT_NAME, "Raspberry Pi 4 Model B"),
+			DMI_EXACT_MATCH(DMI_BOARD_NAME, "Raspberry Pi 4 Model B"),
+		},
+		.driver_data = (void *)&raspberrypi_4b_data,
+	},
 	{}
 };
```

jlinton, 2021-01-19:
I don't think that is needed, you just need a recent firmware repo

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/?id=05789708b79b38eb0f1a20d8449b4eb56d39b39f

gesangtome, 2021-01-20:
@jlinton Helpful

jlinton, 2021-01-20:
This landed today too:

https://git.kernel.org/pub/scm/linux/kernel/git/ulfh/mmc.git/commit/?h=next&id=22510bcd5547f6847c821c12785b4ed435c61662

So the only thing remaining is the edk2 set.

jlinton, 2021-02-22:
The edk portions are merged for the SD/Wifi, so that leaves the BT bits, i'm going to change the title.

gesangtome, 2021-02-23:
@jlinton nice

jlinton, 2022-03-22:
Ok, so the key to bluetooth continues to be that it needs to be running on the PL011 if booted in ACPI mode. This tended to somewhat work until the latest round of uart patches went in and no longer always init the pl011 at 115200. Further, the reliability problems in the past seem like they were partially because I wasn't runtime updating the BT firmware with the files from https://github.com/RPi-Distro/bluez-firmware/tree/master/broadcom. Now that the miniuart patches have landed in mainline+EDK2 it should become possible to boot with that as the console as well.