pftf/RPi4 #120: NVME support for RPi Compute Board 4

CreeperKong, 2020-12-24:
The CM4 provides a PCI Express x1 Slot extension on the IO Board.
Is it possible to give support for NVME on the platform?

jlinton, 2021-01-04:
Yes, we are working in that direction. There are two parts, first the firmware needs an update to deal with the mapping changes the rpi foundation made to the pcie mapping between the original rpi4 and the CM4. At that point we could add the NVMe driver to the firmware, but the OS's will also need to directly support PCIe rather than the XHCI platform device we have in place for compatibility purposes. That is also progressing in the form of a generic ACPI/PCIe SMC conduit standard.

So I would expect that is possible in the near future.

rtreffer, 2021-03-12:
ACPI+devicetree works if the following patch is applied: https://github.com/rtreffer/edk2-platforms/commit/126fdc009f215b3491760011ac08f8515e2b6bfd

@jlinton would you consider the missing ACPI only support a blocker against enabling this upstream?

jlinton, 2021-09-01:
The kernel patches look good (their Acks), I would hope they land in 5.16 at this point, which itself will be a couple of months out. This gives me time if they get merged to -next/etc in the near future, to fix the fact that I changed the _DSD that is now in the .30 firmware just built. <sigh>

For now I would suggest use the .30 release, and merge the linked above patches with the third patch's line 

if (strlen(soc) != ACPI_OEM_ID_SIZE)

changed to:

if (strlen(soc) < ACPI_OEM_ID_SIZE)

and 
"bc2711" changed to "bcm271" in the table.

xvzf, 2023-12-23:
Hey folks

Any update on this?

LiquidPL, 2025-07-09:
As far as I can tell, the relevant patches have already [been](https://github.com/tianocore/edk2-platforms/blob/6153891ee3b4e80e0340236f0eec0228746bacc0/Platform/RaspberryPi/RPi4/RPi4.dsc#L769) [applied](https://github.com/tianocore/edk2-platforms/blob/6153891ee3b4e80e0340236f0eec0228746bacc0/Platform/RaspberryPi/RPi4/RPi4.fdf#L294) to the edk2-platforms tag used by this project.

CreeperKong, 2025-07-18:
> As far as I can tell, the relevant patches have already [been](https://github.com/tianocore/edk2-platforms/blob/6153891ee3b4e80e0340236f0eec0228746bacc0/Platform/RaspberryPi/RPi4/RPi4.dsc#L769) [applied](https://github.com/tianocore/edk2-platforms/blob/6153891ee3b4e80e0340236f0eec0228746bacc0/Platform/RaspberryPi/RPi4/RPi4.fdf#L294) to the edk2-platforms tag used by this project. I figure this can be closed now.

I have tested this on my CM4, it worked now!

closed 2025-07-18 by CreeperKong