pftf/RPi4 #121: Headless Configuration

buckaroogeek, 2020-12-26:
I have recently been working on setting up and running a headless RPi4 (8GB) with Fedora 33 Minimal. This was fairly easy and straightforward with a microSD.  Fedora, which also powers my desktop workstation, provides an arm-image-installer application that does most of the heavy lifting. Updating the EFI partition with files from the current release of this project were the final step needed for success. Once booted, Fedora minimal also reported the correct ram with `free`.

Booting successfully from a USB attached SSD took a lot more effort and investigation over a couple of weeks. The solution was gleaned from an issued thread in this project (which I cannot find at the moment) which noted that success with a USB attached SSD requires the insertion of a blank microSD card. Once I did that, set up and booting from an SSD follows the same steps used for a microSD boot. `free`, however, shows that the 3 GB limit is imposed by the UEFI boot process.

I could not find any way to configure the UEFI boot process to disable the 3 GB limit when setting up a headless system. I finally had to purchase and connect a 7 inch touch screen to the RPI4 and use the UEFI user interface to disable the 3 GB limit. I could also have used the UEFI shell. 

Is there a way to configure settings such as the 3 GB limit with a headless system? If not, would it be possible to devise something suitable? My personal goal is to set up kubernetes on a small RPi4 cluster creating, in effect, an efficient and reliable computing appliance for remote networks like mine support farm operations at the end of a limited wireless connection to the world. Having to plug a video device and keyboard into each RPi4 to disable the 3 GB limit seems like it could be a lot more simple. Have I missed something?  If not, is a headless configuration process feasible?

thanks

Brad

kujeger, 2020-12-27:
I asked around in discord a bit back about configuring the firmware without entering the setup, and it is apparently not possible.

Re. sd cards and booting from usb: I have several pi4s that run entirely off usb sticks with no sd cards attached, but that required updating the eeprom to 2020-09-14 (or newer): https://github.com/raspberrypi/rpi-eeprom

worproject, 2020-12-28:
You can use a serial adapter to configure the UEFI without a screen. The same RPI_EFI.fd file can be then used for all boards, since the variables are stored in the file itself.

buckaroogeek, 2020-12-28:
kujeger - thanks. I have to think that if there is a UI or a shell, then a process supporting headless configuration (e.g. something similar to the config.txt file the raspberry boot loader uses) should also be possible, I can boot raspian OS from a USB but not Fedora which is using grub, uboot and uefi - without the blank microSD that is. If I recall correctly the UEFI boot process slips into a loop testing for the presence of a microSD card.

@worproject - thanks. Good to know that RPI_EFI.fd stores the configuration and can be copied to other RPi4s. I looked at using a serial adapter but since I am using a POE Hat, any that connects to the gpio pins is not feasible without additional hardware.

best regards

Brad

pbatard, 2020-12-28:
> If I recall correctly the UEFI boot process slips into a loop testing for the presence of a microSD card.

That was [fixed in 1.17](https://github.com/pftf/RPi4/releases/tag/v1.17).

buckaroogeek, 2020-12-28:
Pete - thanks for the update. I verified that I am using files from the 1.21 release . The screen I have to use with the RPI4 is small (7 inches) and the text too small to easily read. I can say, that if the microSD slot is empty, the screen displays the rainbow splash, about 5 lines of text, another 6 or so lines, then loops to the rainbow again. I am using the most recent EEPROM too. Not sure what the trigger is but also not concerned about it - a blank microsSD is a small price to pay. Probably user (me) error!

buckaroogeek, 2021-02-14:
Given the comment from @woproject I have a workaround in place.

Thanks!

closed 2021-02-14 by buckaroogeek

netstx, 2021-04-14:
> 
> 
> You can use a serial adapter to configure the UEFI without a screen. The same RPI_EFI.fd file can be then used for all boards, since the variables are stored in the file itself.

Do I have to set anything special on config.txt ? I tried both dtoverlay=miniuart-bt and dtoverlay=disable-bt and could not see any output over serial (booting from USB stick). Booting Raspbian from SD Card works fine on serial.