pftf/RPi4 #122: Keyboard does not respond

pe1msz, 2021-01-03:
During the boot, I get the nice splashscreen, and I press escape multiple times. it does not get me in setup.
Tried 2 different USB-wired keyboards, and 2 pi4's. Do I forget something simple?

Biune, 2021-01-08:
Same issue @pe1msz, it's really annoying.

Digasi, 2021-01-10:
I have the same problem. PI4 8GB models.

zhanghua000, 2021-01-13:
pi 4b 4gb version, also has this problem.

tuxevil, 2021-02-05:
> Pi 4B 4gb version also has this problem. Downgrade to v1.21 can solve this problem.

did anyone fixed with this?
mine still not working

zhanghua000, 2021-02-07:
> > Pi 4B 4gb version also has this problem. Downgrade to v1.21 can solve this problem.
> 
> did anyone fixed with this?
> mine still not working

On my Pi, there is also another solution: use Ubuntu Server image. I can use it normally even with the latest version of firmware. But maybe the causes of our problems are not same. My problem is that my keyboard isn't respond after booting into install environment, everything is normal before booting into Debian install image.

tuxevil, 2021-02-08:
> On my Pi, there is also another solution: use Ubuntu Server image. I can use it normally even with the latest version of firmware. But maybe the causes of our problems are not same. My problem is [...]

Im trying to install EsXI but I cant open bios menu

toro-, 2021-02-15:
I've also experienced this issue. 
Tried both USB2 and 3 ports as well as two different keyboard.
It seems to work on my 2gig Pi 4, but not my 8gig Pi 4 - extremely frustrating as I wish to install ESXi - 2gig is (understandably) not supported by VMware.

tuxevil, 2021-02-16:
> I've also experienced this issue.
> Tried both USB2 and 3 ports as well as two different keyboard.
> It seems to work on my 2gig Pi 4, but not my 8gig Pi 4 - extremely frustrating as I wish to install ESXi - 2gig is (understandably) not supported by VMware.

someone told me to remove "gpu_mem=16" from the config.txt (as many youtubers tells to do) and it worked for me, give it a try

samerhaj, 2021-02-16:
It is very possible that these USB issues on some RPi4 (and Pi400) models are due to https://github.com/raspberrypi/firmware/issues/1518

toro-, 2021-02-17:
> someone told me to remove "gpu_mem=16" from the config.txt (as many youtubers tells to do) and it worked for me, give it a try

Just removed that line from my config.txt. works a treat!

thanks for the info. really appreciate that

tetofonta, 2021-03-03:
Same problem with CM4 =(

samerhaj, 2021-03-03:
Can you please confirm if this issue is fixed in https://github.com/pftf/RPi4/releases/tag/v1.24 ?

tetofonta, 2021-03-04:
I tested v1.24 on CM4 1GB RAM, the keyboard does not work when the splash screen is reached. (I can't enter into setup nor continue booting).

As you can see, i tried with the dwc2 overlay because on CM4 usb ports are disabled by default, but that has not made any change; the keyboard is still ignored.

This is what UART spits out:
```
recover4.elf not found (6)
recovery.elf not found (6)
Read start4.elf bytes  2228800 hnd 0x000010fb hash 'dce1545e7133835b'
Read fixup4.dat bytes     5448 hnd 0x000010eb hash '4109cea6ad32e5fe'
0x00a03140 0x00000000 0x000000ff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0

MESS:00:00:05.113371:0: arasan: arasan_emmc_open
MESS:00:00:05.299473:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.302269:0: brfs: File read: 235 bytes
MESS:00:00:05.396797:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.399558:0: gpioman: gpioman_get_pin_num: pin DISPLAY_SDA not defined
MESS:00:00:05.406469:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:05.428956:0: brfs: File read: 235 bytes
MESS:00:00:05.435659:0: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
MESS:00:00:05.440288:0: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
MESS:00:00:05.448028:0: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
MESS:00:00:05.455478:0: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
MESS:00:00:05.951228:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:05.956866:0: *** Restart logging
MESS:00:00:06.000418:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:06.006353:0: HDMI0: hdmi_pixel_encoding: 300000000
MESS:00:00:06.011828:0: HDMI1: hdmi_pixel_encoding: 300000000
MESS:00:00:06.017940:0: gpioman: gpioman_get_pin_num: pin CAMERA_0_I2C_PORT not defined
MESS:00:00:06.029279:0: dtb_file 'bcm2711-rpi-cm4.dtb'
MESS:00:00:06.036260:0: brfs: File read: /mfs/sd/bcm2711-rpi-cm4.dtb
MESS:00:00:06.039502:0: Loading 'bcm2711-rpi-cm4.dtb' to 0x1f0000 size 0xc032
MESS:00:00:06.058848:0: brfs: File read: 49202 bytes
MESS:00:00:06.126530:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:06.129098:0: brfs: File read: 235 bytes
MESS:00:00:06.134346:0: brfs: File read: /mfs/sd/overlays/miniuart-bt.dtbo
MESS:00:00:06.159850:0: Loaded overlay 'miniuart-bt'
MESS:00:00:06.226050:0: brfs: File read: 1819 bytes
MESS:00:00:06.228737:0: brfs: File read: /mfs/sd/overlays/dwc2.dtbo
MESS:00:00:06.248297:0: Loaded overlay 'dwc2'
MESS:00:00:06.249575:0: dtparam: dr_mode=host
MESS:00:00:06.298236:0: brfs: File read: 801 bytes
MESS:00:00:06.300590:0: Failed to open command line file 'cmdline.txt'
MESS:00:00:07.523007:0: brfs: File read: /mfs/sd/RPI_EFI.fd
MESS:00:00:07.525467:0: Loading 'RPI_EFI.fd' to 0x0 size 0x1f0000
MESS:00:00:07.531292:0: No compatible kernel found
MESS:00:00:07.535793:0: Device tree loaded to 0x1f0000 (size 0xc5d4)
MESS:00:00:07.543459:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:07.550950:0: uart: Baud rate change done...
MESS:00:00:07.552967:0: uart: Baud rate change done...
MESS:00:00:07.558145:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
NOTICE:  BL31: v2.3():v2.3
NOTICE:  BL31: Built : 10:40:51, Apr 21 2020
UEFI firmware (version UEFI Firmware v1.24 built at 19:31:21 on Feb 26 2021)
[2J[04D[=3h[2J[09DESC (setup), F1 (shell), ENTER (boot)......
```

rtreffer, 2021-03-11:
There is currently a bug in the EDK2 RPI platform that prevents the USB 2.0 port to be initialized.

I have working USB 2.0 on the CM4 after applying the following patch: https://github.com/rtreffer/edk2-platforms/commit/ca053c606c216a0d90239e5fdcbd5118aa0660d6

I'll try to upstream that now.....

rtreffer, 2021-03-11:
I've uploaded a test release to https://github.com/rtreffer/RPi4/releases/tag/dwc2-reset

Please let me know if it fixes the issue.

samerhaj, 2021-03-11:
This may be related to what was reported here: https://github.com/pftf/RPi4/issues/88#issuecomment-792370668

The patch in rtreffer/edk2-platforms@ca053c6 is consistent with the expected behavior from the change in https://github.com/tianocore/edk2-platforms/commit/f8958b86e8863432b815a132a0f0fe82950c6dd1

ljfranklin, 2021-03-12:
@samerhaj yep, seems like https://github.com/rtreffer/edk2-platforms/commit/ca053c6 also allows iPXE to correctly initialize network devices on the Raspberry Pi.

In the other thread you asked whether I was using a USB NIC, no just the built-in Pi ethernet. But iPXE seems to use this USB ethernet driver to control the built-in Pi ethernet: https://github.com/ipxe/ipxe/blob/02280dc642907b908f4b5c7e0d82d8ad1d51d574/src/drivers/net/lan78xx.c#L36.

samerhaj, 2021-03-12:
Thanks @rtreffer  and @ljfranklin .

Were others able to confirm that https://github.com/rtreffer/RPi4/releases/tag/dwc2-reset fixes the USB keyboard issue?

tetofonta, 2021-03-12:
@samerhaj Yay, tested it. keyboard works on CM4

samerhaj, 2021-03-12:
Fix committed upstream https://github.com/tianocore/edk2-platforms/commit/2620e05c6fade20ef5e1ba69280d09f9e2b3eff6

pbatard, 2021-03-25:
This should be fixed in the latest release.

closed 2021-03-25 by pbatard