pftf/RPi4 #123: Overflow in XHCI-HCD while booting RHEL with version > 1.20

andreamtp, 2021-01-08:
On a RPi 4b 8Gb I've succesfully installed rpi4-uefi on an USB stick (limiting the visible ram to 3gb) where is also installed [RHEL8 for ARM](https://developers.redhat.com/products/rhel/download).

After upgrading rpi4-uefi to 1.21 the boot started to fail due to an error on xhci-hcd driver:

`xhci-hcd PNP0D10:00: DMA addr 0x00000000a3930000+65536 overflow (mask fffffffffffffffffff, bus limit 7fffffff)`

[image]

The same problem happen with 1.22, with pretty equal message:

[image]

After reading 1.22 release notes, I've tried to use 1.20 start4.elf and fixup4.dat with 1.22 rpi4-uefi, with the same results.

The only way to let the kernel operate correctly is to downgrade to 1.20.

Please let me know if you need further informations or what kind of further data you may need.

thanks

jlinton, 2021-01-12:
Which RHEL version is this? I'm guessing there is an interaction with the IORT table changes in 1.20.

andreamtp, 2021-01-13:
Il 12/01/21 19:00, Jeremy Linton ha scritto:
>
> Which RHEL version is this? I'm guessing there is an interaction with 
> the IORT table changes in 1.20.
>
[root@pi ~]# cat /etc/redhat-release
Red Hat Enterprise Linux release 8.3 (Ootpa)
[root@pi ~]# uname -a
Linux pi 4.18.0-240.10.1.el8_3.aarch64 #1 SMP Wed Dec 16 03:21:37 EST 
2020 aarch64 aarch64 aarch64 GNU/Linux

jlinton, 2021-01-22:
Thanks, 

Without testing this. I think the fix is probably to tie the creation of the IORT table to the 3G limit flag, so that it only appears when the 3G limit is lifted.

jlinton, 2021-01-28:
Ok, I just validated this, its definitely the IORT. The fix is pretty trivial for the current code:

something like:
`  case SIGNATURE_32 ('I', 'O', 'R', 'T'):
    // only enable the IORT on machines with >3G and no limit
	// to avoid problems with rhel/centos
    if (PcdGet32 (PcdRamLimitTo3GB)) {
      return FALSE;
    }
`
In the ConfixDxe.c file.  

Except, I've gone and lowered the limit in the IORT from 2G to 1G. Which means there needs to be a way to disable it on 2G rpi4's now too. So its a bit more complex, we want to disable it for machines with the 3G limit or 2G machines with the emmc2 disabled. Bit tricky there, as 2G users suddenly have an extra step. What really needs to happen is the 3G limit flag can be enabled on 2G machines, making it more a "Legacy memory option".

jlinton, 2021-02-01:
https://edk2.groups.io/g/devel/topic/bug_patch/80300339?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,80300339

pbatard, 2021-02-25:
This should be fixed in the 1.23 release.

closed 2021-02-25 by pbatard

andreamtp, 2021-03-01:
Hi team, I can confirm that 1.24 is working fine as expected with RHEL 8. 
Thanks