pftf/RPi4 #125: No more iSCSI boot on 1.22 ?

detonyle, 2021-01-13:
Just upgrade from 1.20 to 1.22 and iSCSI option isn't available anymore.
(still present in 1.21)

pbatard, 2021-01-14:
This is a result of [this edk2 commit](https://github.com/tianocore/edk2/commit/4ecb1ba5efa3fcce850aa979dd80ffa3d322f976), which disabled iSCSI by default for the following reason:

> iSCSI is using the deprecated function MD5. It is
> better to make the default setting secure. If the platforms
> want to use the iSCSI, they should enable it in the platforms'
> dsc file and be aware they are using an function with weak
> cryptography.

I guess that, since the Raspberry Pi platform can't exactly be qualified as a "secure" platform in the first place, we'll try to reenable the feature in a future firmware release.

detonyle, 2021-01-14:
Thanks Pete for feedback! yeah exactly secure isn't primary quality of RPi, I have few RPi4 for some ESX-ARM testing and iSCSI boot is quiet nice, so I stay on 1.21 for now, even if sometime I get SDCard corruption on some reboot

mry, 2021-02-06:
Having suffered from the corruption bug in combination with a three story house, I forked and applied the patch to re-enable iSCSI on the latest EDK2 release on the v1.22 release. 
It's faster for @pbatard to apply the fix directly rather than me doing a pull request.

https://github.com/mry/RPi4/releases

pbatard, 2021-02-06:
Note that we don't apply Pull Request related to submodules to this repository.

The patch to re-enable iSCSI is something that needs to be applied to the `edk2-platforms` submodules, and therefore it will not be accepted here, even if you create a Pull Request.

If there's any behaviour that you want, and that we are not addressing ourselves in the timeframe you expect, you are strongly encouraged to submit a patch to the [edk2-devel mailing list](https://edk2.groups.io/g/devel), which is where any patch that you want to see applied to the RPI4 firmware must go through.

pbatard, 2021-02-08:
Actually, my comment above was incorrect, since this is one of the rare case where we don't need to send a patch to EDK2 but can directly toggle a build flag in [`build_firmware.sh`](https://github.com/pftf/RPi4/blob/master/build_firmware.sh). I'll make sure to have the iSCSI build flag set for the next release.

samerhaj, 2021-02-22:
I think this is as easy as overriding NETWORK_ISCSI_ENABLE in RPI4.dsc. This is indeed what was suggested in the commit that turned off iSCSI by default: https://github.com/tianocore/edk2/commit/4ecb1ba5efa3fcce850aa979dd80ffa3d322f976

> iSCSI is using the deprecated function MD5. It is
> better to make the default setting secure. If the platforms
> want to use the iSCSI, they should enable it in the platforms'
> dsc file and be aware they are using an function with weak
> cryptography.

samerhaj, 2021-02-22:
Closed by mistake

pbatard, 2021-02-22:
Yes, please don't close this issue. I have the patch ready, which will close this issue when pushed. But I don't want to push it before we start a new firmware build, which is conditioned by the Pi Foundation officially updating their firmware (per raspberrypi/firmware#1518).

samerhaj, 2021-02-22:
I think we should simply turn on NETWORK_ISCSI_ENABLE in RPI4.dsc and submit a patch to tianocore, rather than overriding it in the build script. This is exactly what was intended of this commit (tianocore/edk2@4ecb1ba)

pbatard, 2021-02-22:
That had been my initial plan, but I dropped it because I don't see the point when my take is that people who build the firmware and need iSCSI probably need to be made aware, with regards to the other platforms they might be using, that iSCSI has been dropped from being included by default due to using insecure protocols.

Granted, with the Pi platforms being non-secure by nature, it doesn't make much difference for us where we add it. But in general, you want to go with the best practices in the `.dsc`, and if the current best practice is that iSCSI should be manually added as a build parameter, to bring attention that it introduces security risks, that's what I would advocate we follow too.

Now, I don't actually have a bone where we do it, and if there's a majority that wants it in the `.dsc` I don't see an issue with a patch being submitted for that. But, right now, I don't think this is the best approach.

commit 1367aefb (2021-02-25): Add iSCSI support back

* EDK2 commit tianocore/edk2@4ecb1ba5efa3fcce850aa979dd80ffa3d322f976 removed iSCSI support
* Add it back through the corresponding build option, and use this opportunity
  to factorize the build flags and have DEBUG and RELEASE use the same ones.
* Closes #125
closed 2021-02-25 by pbatard