pftf/RPi4 #128: Failed to boot Linux Kernel using GRUB2

zhanghua000, 2021-02-08:
I am using Manjaro-ARM on my Raspberry Pi 4B. I can boot into GRUB2 successfully, and everything looks like working. But after I chose menu entry to start my os kernel, the Raspberry Pi is not respond after loaded initramfs. If I run commands manually in GRUB Shell to boot the same Kernel, after I run `boot` command, the screen shows the follows and the GRUB Shell will be stuck:
```
EFI stub: Booting Linux Kernel...
EFI stub: Using DTB from configuration table
EFI stub: Exiting boot services and installing virtual address map...
```
The GRUB is installed from ArchlinuxARM's repository. I have tried using official Kernel and the Kernel compiled by myself, both are the same result. I have also tried to use GRUB on my Raspberry Pi to boot Ubuntu Server installation image, it works correctly even I directly copy its Kernel and initramfs and boot them by using GRUB installed on my Raspberry Pi. Is it because the default Raspberry Pi's Kernel config should be modified? Or I should recompile GRUB with special config?  Or it is because the bug of this firmware? I have googled this problem but there is not too much information about this. Thanks for your help in advance.

baryluk, 2021-02-11:
Which version of Manjaro, and what kernel version? Also do you use ACPI or DT+ACPI?

zhanghua000, 2021-02-16:
I am using Manjaro-ARM latest version. The official Kernel is provided by package `linux-rpi4-mainline` and its version is 5.10.14, the Kernel compiled by myself is provided by the `rpi-5.10.y` branch of https://github.com/raspberrypi/linux repository, its version is also 5.10.14. The build config is `bcm2711_defconfig` and switch the frequency governor to ondemand (from the default, powersave).

baryluk, 2021-02-16:
@zhanghua000 You shouldn't be using rpi4 kernel package I think, just use generic arm package with UEFI.

Also I am pretty sure the `rpi-5.10.y` from raspberrypi repo, has same issue, it has no UEFI or ACPI, etc. You need to add it up in config manualy on top of `bcm2711_defconfig`.

jlinton, 2021-02-16:
You can also use earlycon (depends on a few things you might need to provide the full earlycon=pl011,mmio32,0xfe201000 or earlycon=uart8250,mmio32,0xfe215040) to see why the kernel is failing.

That said, by default this firmware is intended for ACPI kernels, but you can enable a DT in the rpi advanced menus. It will then pass the dtb+overlays provided to the GPU start4.elf/etc firmware up the stack to the kernel.

zhanghua000, 2021-02-17:
I have tried to compile generic kernel and I can successfully boot into Kernel. I think rpi4 kernel may provide some special code for its hardware, so I use rpi4 kernel.  But now it looks like I think too much about it. Thanks for everyone's help.

closed 2021-02-17 by zhanghua000