pftf/RPi4 #130: I2C driver not discovered on Linux

andysan, 2021-02-16:
I2C is currently not supported on Linux when booting using UEFI. The following issues need to be resolved before the patches can be posted:

- There is currently no way of distinguishing between the BCM2835 and BCM2711 I2C blocks based on information in the ACPI tables. The driver needs to know the revision to turn on the right quirks.
- Linux currently doesn't know how to interpret honour pin function descriptors. This means that the firmware needs to configure the pin mux to route the I2C signals before handing over to Linux. I have a patch that adds this functionality.
- There is no (obvious) way of discovering the frequency of the clock driving the I2C block. The current driver relies on the clocks being declared in the device tree.

I have a series patches that "work", but it relies on a hard-coded clock frequency and assumes it is running on a BCM2711. That's not ideal and won't be good enough for mainline.

We should also consider whether we want the I2C buses to be enabled by default or if we want to make the pin functions configurable. For example, the I2C bus used for hat identification (currently not exported) can be used as a serial port instead on the BCM2711.

andysan, 2021-02-18:
Having talked to some people who actually understand ACPI (my understanding is currently a bit limited, it seems like the cleanest solution to the problems above is to let the firmware configure the pinmux and the I2C bus frequency. My impression is that the _HRV field could be used to identify the revision of the IP block and enable the relevant quirks.

It seems like this implies that we need a general HAT configuration mechanism in UEFI. Some hats have configuration EEPROMs that could be used to automatically configure the pinmux and expose the right buses to the OS.

aBradleyUno, 2022-03-30:
Has there been any progress with this?