pftf/RPi4 #131: Persistent UEFI settings across reboots while netbooting on RPI4

Bobberty, 2021-02-18:
Not too sure if this has been covered here.
I have a netboot infrastructure with a mix of iPXE and DNSMASQ.
The RPI4 boots fine, but needs to be configured to boot off of the UEFI IPv4 after every reboot.
Further, none of the UEFI settings are kept between reboots.
This RPI4 does not have an SD card inserted while booting.

samerhaj, 2021-04-12:
The configuration settings are stored in the RPI_EFI.fd file in the SD card. Without an SD card, you will be using the default settings on every boot. Did you try to first use an SD card, configure the settings and save then, then store a copy of the RPI_EFI.fd (that contains the updated settings) on the network boot server?

Bobberty, 2021-05-06:
In a perfect world, I would like to deploy RPIs without an SD card.  Or do an install onto a bare SD card.  
As a solid untested/unknown suggestion, allow the RPI_EFI.fd to be pulled from the same TFTP server as the binary.  Possibly referenced by MAC some pxe solutions work.    Writing back may be difficult, but each config may be shared or modified as required.

OGKevin, 2021-05-20:
Ah similar to #151

Bobberty, 2021-06-01:
But, the difference would be that the boot order would be pulled from the TFTP unless there is no network available.  Only then refer to the on board RPI_EFI.fd.