pftf/RPi4 #132: Raspberry PI 400 USB/ Keyboard does not work

nsa666666, 2021-02-19:
Related to https://github.com/pftf/RPi4/issues/109 

PI 400 Boots with recent version, but USB and integrated keyboard does not work in Setup and in OS after boot.

nsa666666, 2021-02-19:
This seems to fix the issue https://github.com/raspberrypi/firmware/issues/1518#issuecomment-779355320 tough.

jeroenbrons, 2021-02-25:
same for me, at least in setup. the current firmware from the Hexxeh repo works in UEFI-setup, currently booting Win10 to test if USB works when booted in a OS

pbatard, 2021-02-25:
Could this be related to #122? In other words, are you sure you don't have a `gpu_mem=` line in your `config.txt`?

samerhaj, 2021-03-11:
You could also look at: https://github.com/pftf/RPi4/issues/122#issuecomment-796580190 , and try that FW binary or change to see if it fixes the issue

samerhaj, 2021-03-12:
Were you able to confirm if https://github.com/rtreffer/RPi4/releases/tag/dwc2-reset fixes the USB keyboard issue?

jlinton, 2021-04-08:
So the keyboard is attached to the DWC2 controller. If so you will need:

https://lkml.org/lkml/2021/4/7/1725
and
https://edk2.groups.io/g/devel/topic/patch_3_3/81935645?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,81935645

Although, its possible that second patch is in error, and we need to fix linux instead.

samerhaj, 2021-04-16:
> So the keyboard is attached to the DWC2 controller. If so you will need:
>
> https://lkml.org/lkml/2021/4/7/1725
> and [...]

Confirmed that the EDK2 patch above is not in error. The ACPI spec has a typo in the ASL example, and need to be corrected. This is being tracked in this ACPI spec ECR: https://bugzilla.tianocore.org/show_bug.cgi?id=3335

jlinton, 2021-05-19:
The next version of the firmware should have this fixed.