pftf/RPi4 #135: dual monitors

andrewufrank, 2021-02-26:
The Raspian OS and the ubuntu 20 installation makes use of the dual monitor capability. I tried to install Debian with the netbios bullseye downloaded (following the instructions from [akeo](https://www.raspberrypi.org/forums/viewtopic.php?f=50&t=282839#p1713105) and then installed Debian Mate, expecting to use the dual monitors (as I do with the Ubuntu installation). Unfortunately, only a single monitor is currently usable. 
There are perhaps some tweaks missing (dtoverlay=vc4-fkms-v3d ? but where to include?) or some more software? Any suggestions and help would be appreciated - the dual monitor capability of the RPi4 is a novel feature for a SBC and a regular Debian installation should be capable of using it?
Thank you!

andrewufrank, 2021-02-27:
I have added now the dtoverlay=vc4-kms-v3d to the config.txt and moved the the vc4-kms-v3d.dtbo into the overlays folder in /boot/efi. xrand does not show the hdmi monitors (just screen). what else is missing?

jlinton, 2021-03-03:
You might note: https://github.com/pftf/RPi4/issues/127

kpinc, 2025-03-08:
I happened to notice this issue.  I also have this problem with stock debian (12,  bullseye).  The 2nd monitor sits there and retains "the rainbow screen". 

"The Internet" says to try adding the following to config.txt:

hdmi_edid_file:1=1
hdmi_edid_filename:1=edid.dat
hdmi_force_hotplug:1=1

When I do this, the screens go blank.  (It feels like a crash, but I don't know what's going on under the covers.)

My guess was that because I've only 3G of usable RAM (on a rpi4B w. 8G) that that was causing the problem, and I worked at trying to solve that.  See issue #265 .

I have not tried any of the solutions proposed.  I have upgraded the eeprom firmware to the latest version and that didn't fix anything.  And I'm running 1.41 of the UEFI firmware.

andrewufrank, 2025-03-08:
I am away from my RPi installation and cannot test. I had end 2024 a dual monitor installation running (with the 8GB RPi4) and think it was a stock installation of debian from the inofficial debian for raspi ["tested images](https://raspi.debian.net/tested-images/l) with bookworm. 
I have since migrated to a RPi5 and use the  [armbian](https://www.armbian.com/rpi4b/l) minimal IOT image and install then the mint desktop with `apt install`. There are not yet pure debian images for the RPi5 on the inofficial debian site. 
Dual monitor work well, but I cannot currently test. I notice a noticable difference in web browsing between RPi4 and RPi5 (i.e. RPi5 is usable for web browsing...), not so much on other programs.

kpinc, 2025-03-08:
On Sat, 08 Mar 2025 00:42:24 -0800
andrew u frank ***@***.***> wrote:

> andrewufrank left a comment (pftf/RPi4#135)
> 
> I am away from my RPi installation and cannot test. I had end 2024 a
> dual monitor installation running (with the 8GB RPi4) and think it
> was a stock installation of debian from the inofficial debian for
> raspi ["tested images](https://raspi.debian.net/tested-images/l) with
> bookworm.

Note that I installed debian using the debian installer, not from
an image.  Onto a SSD.  On that install I can't use 2 monitors
(or, seemingly, tell UEFI to let me use more than 3G).

The debian image (bullseye, iirc) from debian.org (onto CF) gets me 8G
and I can use 2 monitors.