pftf/RPi4 #137: Linux kernels cannot load network: (failed to connect to PHY) when UEFI netbooted through iPXE or grub

wernerb, 2021-03-07:
Linux kernels started with network adapter already loaded previously fail to connect in final boot. This is on kernel 5.10.19.

I tested both grub2 and iPXE to make sure there is not a problem there.

Both will boot flatcar linux/fedora coreos over the internet but it shows `failed to connect to PHY` and `Unable to find mii`.
Reading the code a little at the linux kernel it seems it can't find the mac adress it already initialized. I also see when booting some warnings about ACPI table being corrupted. I don't know if this is related.

OS tried:
* OS Fedora Coreos (33.20210127.dev.0) https://fedorapeople.org/groups/fcos-images/builds/latest/aarch64/
* Flatcar (latest alpha)  https://alpha.release.flatcar-linux.net/arm64-usr/current/

I verified that both can be installed locally directly without internet through usb stick and then booted correctly with no ethernet issues afterwards. It really seems re-use of the adapter that gives problems.

I should note that I am also booting the firmware from tftp instead of sd card.

This is a debug log of RPI4 master built with debug and the lastest zip of firmware files.
https://gist.github.com/wernerb/3d195cfdb640ec66d9029197e3be91b8 (Using

**Reproduce with netboot.xyz**

boot `http://boot.netboot.xyz/ipxe/netboot.xyz-arm64.efi` manually through the Network boot. 
Then when booted into the menu.
Select `Execute iPXE shell`

Type in the following:

```
kernel https://alpha.release.flatcar-linux.net/arm64-usr/current/flatcar_production_pxe.vmlinuz initrd=flatcar_production_pxe_image.cpio.gz flatcar.config.url=http://urldoesntmatter flatcar.first_boot=yes console=tty0 console=ttyS0 flatcar.autologin
initrd https://alpha.release.flatcar-linux.net/arm64-usr/current/flatcar_production_pxe_image.cpio.gz
boot
```

You will see messages such as:
```
[  133.316150] Unable to find mii
[  133.319239] bcmgenet BCM6E4E:00 enabcm6e4ei0: failed to connect to PHY
```

jlinton, 2021-03-09:
This is because your failing to load the initrd, which contains the mdio_bcm_unimac.ko which is the mdio/phy driver for the genet.

Its likely related to https://github.com/pftf/RPi4/issues/113

noahwilliamsson, 2021-03-28:
@wernerb FWIW, I just reported the apparent lack of CONFIG_MDIO_BCM_UNIMAC in the Fedora 34 pxeboot images.

I've found that:

- http://ftp.acc.umu.se/mirror/fedora/linux/releases/test/34_Beta/Server/aarch64/os/images/pxeboot/ currently ships 5.11.3 without CONFIG_MDIO_BCM_UNIMAC, and without patches required for SD and WiFi support
- http://ftp.acc.umu.se/mirror/fedora/linux/development/34/Server/aarch64/os/images/pxeboot/ currently ships 5.11.10 without CONFIG_MDIO_BCM_UNIMAC but with patches required for SD and WiFi(?) support
- http://ftp.acc.umu.se/mirror/fedora/linux/development/rawhide/Server/aarch64/os/images/pxeboot/ currently ships 5.12.0-0.rc4.20210325, which has both CONFIG_MDIO_BCM_UNIMAC and support SD and WiFi

Regarding the SD and WiFi support, I believe the patches in 5.12 that d9e569d0311a27b3ed8dcaa62e94645502dec9f3 refers to are the ones available in:

- https://cdn.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.11.7
- https://cdn.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.10.24

Specifically:

- mmc: sdhci-iproc: Add ACPI bindings for the RPi
- mmc: sdhci: Update firmware interface API

wernerb, 2021-03-29:
@noahwilliamsson thank you for this information. I found out that Ubuntu, for example, could boot so I was already looking at kernels. I have asked Flatcar to also add the relevant flag here: https://github.com/kinvolk/Flatcar/issues/375 

For Fedora Coreos, looking at your bug report I hope they backport this to at least dracut-51 which is what is used now in the Rawhide branch at Fedora CoreOS: https://builds.coreos.fedoraproject.org/browser?stream=rawhide .

Because its confirmed this is not a firmeware issue I am closing this. Thanks again @noahwilliamsson

closed 2021-03-29 by wernerb