pftf/RPi4 #138: Is it possible to save/restore the UEFI configuration

offsides, 2021-03-09:
It appears that all the configuration in the UEFI setup is saved in the RPI_EFI.fd file, which is replaced any time the UEFI firmware is upgraded.  Is it possible to save that data to a separate file, and either manually or (even better) automatically restore it when the firmware is upgraded?  Otherwise I have to go into the UEFI setup and turn off the 3GB limit every time I upgrade the UEFI firmware.

Thanks!

jlinton, 2021-03-30:
In my mind the solution to this will be the same as the solution to #102.  That said, there has been some talk of actually taking over a bit of the SPI flash for them as well, but AFAIK, we need to get the pi foundation on board with that.

d3vi1, 2021-05-05:
Yes, read the dmpstore UEFI Shell command manual. It can dump variables to a text file and import them from there.

samerhaj, 2021-07-17:
This will no longer be an issue if we save the UEFI variables in SPI EEPROM (#6 )

rgl, 2023-05-19:
@samerhaj, also having a way to set them in a file would be quite liberating, because we could pre-prepare the sdcard in advance with the required variables (as-in #210).