pftf/RPi4 #139: iSCSI option / re-enable feature request

ppapajohn, 2021-03-20:
Hi all, 

first of all congrats on the amazing stuff you are performing. :) 

Would it be possible to re-enable the iSCSI boot option in v1.23? Without taking too much of your time, I have an rpi4 8GB, and a Synology to serve as an iSCSI server. I intentionally use UEFI firmware v1.21, in order to boot from iSCSI. This firmware throws synchronous exceptions, which are solved in v1.22. The latter though does not allow for iSCSI boot option.

In advance thank you very much for your time and consideration on my request. 

Best Regards,
Panos

ppapajohn, 2021-03-22:
Hi all,
Sorry for missing that you have added back the iSCSI option in v1.24. Please close this feature request, delete it, whatever you deem most appropriate.

Thank you and sorry for the hustle; I was dizzy I guess. 🙂 

Best Regards,
Panos

samerhaj, 2021-03-22:
Closing as fixed in 1.24

closed 2021-03-22 by samerhaj