pftf/RPi4 #141: Ubuntu 21.04 generic kernel (5.11) not booting anymore (Panic)

ofthesun9, 2021-03-31:
Hello,
I want to boot ubuntu using grub, and I am doing this successfully on my Raspberry Pi4 4Gb since few months, using the UEFI firmware (currently v1.24). UEFI is intalled on the SD card (acpi only, 3GB limit disabled). Ubuntu is installed on an usb external drive.

I was using the generic kernel provided by ubuntu 20.10 (kernel 5.8) : booting Ok. Ethernet Ok.
I recently switched to the upcoming ubuntu 21.04, hoping that the kernel 5.11 or 5.12 would bring bluetooth, wifi, and sd card (as soon as the patches from @jlinton are included.

kernel 5.11 provided by linux-image-5.11.0-11-generic (version 5.11.0-11.12) is booting Ok. Ethernet Ok.

Patches from @jlinton are not included in linux-image-5.11.0-11-generic, but they are on linux-image-5.11.0-13-generic (version 5.11.0-13.14), so I was looking forward to install it to get access to the sd card, but this one is not booting anymore:
[image]

jlinton, 2021-03-31:
That FB dev came up a lot sooner than it does on my machine. Mine is still decompressing the initrd at .5 seconds (hello lots of debug garbage). 

Anyway, I guess that you don't have a serial port, but you have a network connection?

You might try initcall_blacklist=efifb_driver_init on the kernel command line to disable the efi framebuffer. That might allow it to boot up enough that you can log in via ssh and dump the dmesg buffer and attach it here.

Also make sure you don't have DMA enabled on the mmc.

Also, for grins try it with the 3G limit enabled. 

I will spin up a -stable kernel, but AFAIK they are working in fedora at the moment since people have been reporting other issues. AKA you should ping the ubuntu people about this too.

Also, BT likely won't work, only wifi+sd.

ofthesun9, 2021-03-31:
I upgraded to 1.25, 3GB limit enabled, PIO mode for sdcard.
I added initcall_blacklist=efifb_driver_init on the kernel command line , and bravo ! I have now wifi and sd card !
here is the dmesg:
[dmesg.txt](https://github.com/pftf/RPi4/files/6239485/dmesg.txt)

Ps: regarding the timeline in the screenshot, I took the snapshot when I was trying to boot with a kernel that I compiled myself from ubuntu sources. I had the feeling that the error was the same with the official ubuntu kernel, so I did not take another snapshot. If needed, I can provide a snapshot of the error with the official ubuntu kernel.

jlinton, 2021-03-31:
So if you boot with the initcall_blacklist removed it still crashes?

ofthesun9, 2021-04-01:
Yes, If I boot with the initcall_blacklist removed, it still crashes
With initcall_blacklist in the cmdline, it will boot properly.

jlinton, 2021-04-02:
Hmm so I just booted the -stable linux-5.11.y branch. which is displaying as 5.11.11 on the CM4 i've been testing with after merging the PCI/SMC patches. So i'm a little off in the weeds WRT to all these official releases.

So two things to try. If you have been building your own kernel, clone git://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git, checkout linux-5.11.y, copy your ubuntu .config file, build, install and boot it. If that fails we know it isn't a patch that ubuntu is carrying. 

Also while your at it, attach your .config file here to and I will build it (thats going to be faster for me since i'm to lazy to boot ubuntu)

jlinton, 2021-04-02:
Also, I keep thinking about this, are you sure your power supply is good? A lot of the fedora problems with the rpi3 were caused by bad power supplies.

ofthesun9, 2021-04-02:
I looked at the recent differences in ubuntu kernel sources, and I think my issue is caused by this:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1914411 (I made a comment there)
I recompiled 5.11 (the ubuntu "version", 5.11.0-13.14) without this patch and it boots ok.

jlinton, 2021-04-06:
Ok, sounds good, lets close this then.

closed 2021-04-06 by jlinton