pftf/RPi4 #143: Add support for custom virtual screen resolution and support for the official Raspberry Pi touchscreen.

CE1CECL, 2021-04-08:
Hi
i need support for a 800x480 resolution on the official Raspberry Pi touchscreen
while 640x480 works, its too small of a res.
I also recommend having a custom resolution support because config.txt is ignored on the touchscreen and I can't find any option to change the res on the touchscreen anywhere, I just see HDMI options and this isn't HDMI.
Thanks!

TheMindVirus, 2021-04-08:
There is already a Native Resolution option in the UEFI and this should already be picking up on changes to config.txt,
providing that you don't have a virtual resolution also checked and your custom timings are added to the text file.

This Instructables article is for setting up that screen (despite the name of the link):
https://www.instructables.com/Setting-Up-an-800X400-5inch-HDMI-LCD-for-Raspberry/
These are their timings:
```
hdmi_group=2
#hdmi_mode=1
hdmi_mode=87
hdmi_cvt=800 480 60 6 0 0 0
```
The extra 3 zero's are not usually required.

Also, if you don't mind testing experimental drivers, I have work-in-progress support for dual-display for UEFI and Windows.
Please take sensible precautions before trying experimental drivers as it can yield some unexpected results.
Modded Driver: [pi4bdd-800x480-WIP.zip](https://github.com/pftf/RPi4/files/6280169/pi4bdd-800x480-WIP.zip)
1080p Source: https://github.com/TheMindVirus/pi4bdd

CE1CECL, 2021-04-08:
Hi there
It still defaults to 800x600 possibly because it’s not an hdmi device .
I also took a look at the modded driver and it blacks out the screen and the I plug in a hdmi tv in and it seems like it shows as second so I’m not sure.

TheMindVirus, 2021-04-08:
Oh nice, glad to hear it's working as originally intended.

I've only told it to find display number 2 out of a possible 5, I think it just happens to be HDMI0 and HDMI1.

The other 3 outputs (composite, DPI and DSI) are similarly connected to PixelValves but I'm not sure if they are counted along with HDMI displays. I don't have any non-HDMI screens I can test with as such...

closed 2025-04-06 by CE1CECL