pftf/RPi4 #144: Modified .fd file results in failed Windows ARM booting via boot order

0n0w1c, 2021-04-10:
I use grub2 to dual boot linux and Windows ARM on a RPi4 with a USB SSD on a powered hub.
If I use the default RPI_EFI.fd, booting works as expected for both Linux and Windows ARM.

However, if I use Setup to change either the 3GB Limit to Disabled, or set ACPI + Devicetree and Save and Reset, and then allow the UEFI to find the boot device via the boot order, Windows 10 ARM will fail to boot:

error: not a valid root device.

Below is the snippit from my grub.cfg file:

menuentry "Windows 10 ARM" {
          echo "Windows 10 ARM booting..."
          insmod part_gpt
          insmod fat
          insmod chain
          set root=(hd0,gpt1)
          chainloader (hd0,gpt1)/EFI/Microsoft/Boot/bootmgfw.efi
}

If I restore the RPI_EFI.fd file, booting works again.

Or if I enter Setup, and use the Boot Manager to select UEFI USB device 2,
it always works.

RPi4_UEFI_Firmware_v1.25

0n0w1c, 2021-04-11:
If you boot with the default .fd in place, enter Setup and then select Reset, this will also produces the same boot failure on the subsequent boot and the .fd must be restored.

RafaelMarinheiro, 2021-04-11:
I'm experiencing something similar with my iPXE build. The trigger is the same: open Setup menu and then select Reset. The symptom I see is that iPXE fails to detect any network devices.

I bisected the RPI_EFI.fd from the pftf releases and it appears that the problem was introduced between 1.13 and 1.14. These are the ones I tested:

- 1.13: OK :heavy_check_mark:
- 1.14: FAIL :x: 
- 1.15: FAIL :x: 
- 1.16: FAIL :x: 
- 1.19: FAIL :x: 
- 1.25: FAIL :x: 

The only commit in this repo was updating the pointers to edk2 and edk2-platforms. Here's the list of commits between these releases:

- edk2: https://github.com/tianocore/edk2/compare/ceacd9e992cd12f3c07ae1a28a75a6b8750718aa...6ff7c838d09224dd4e4c9b5b93152d8db1b19740
- edk2-platforms: https://github.com/tianocore/edk2-platforms/compare/3698b912301b51df5a6823669edbeba72ce05f69...2d07a49e45326a56354167061390deba526b3c70

RafaelMarinheiro, 2021-04-11:
After bisecting it seems like the issue is with https://github.com/tianocore/edk2-platforms/commit/c8000ecccc83b728baf04ced2fedb870bc3bc1b3. Using https://github.com/tianocore/edk2/commit/6ff7c838d09224dd4e4c9b5b93152d8db1b19740 alongside https://github.com/tianocore/edk2-platforms/commit/1c31eea8a3da6fee715200540c4425ca36007bea works, but using https://github.com/tianocore/edk2/commit/6ff7c838d09224dd4e4c9b5b93152d8db1b19740 with https://github.com/tianocore/edk2-platforms/commit/c8000ecccc83b728baf04ced2fedb870bc3bc1b3 doesn't.

RafaelMarinheiro, 2021-04-11:
It seems that devices are not being initialized correctly when you don't open the UEFI Shell or the UIApp. If you do and then you choose to "continue" then iPXE works correctly.

For some reason, however, every boot before opening the UEFI UI App works just fine.

RafaelMarinheiro, 2021-04-11:
Just guessing (with the caveat that I know barely anything about these things): I guess that when you boot before all devices have been enumerated and a boot order has been written to disk then the UEFI needs to initialize all of the devices in order to actually decide what to do.

When a boot order has already been written to disk then only the devices needed for that specific boot option are initialized. This is a problem for EFI images like iPXE that rely on SNP in order to work properly.

However by opening the UI app you need to initialize all devices in order to allow the user to change the boot order. This is why booting into the UI app and then selecting "continue" makes things work.

A confirmation of this theory is that changing the boot order to make PXE boot to happen before iPXE makes iPXE work. This is because PXE boot probably explicitly requires the network devices to be properly initialized. Having it go first initializes the network devices and then iPXE works just fine.

An easy fix for this would be to re-add the the ConnectAll call. I don't know how a "proper fix" would happen since it looks like an arbitrary EFI image could rely on arbitrary devices to be initialized. Perhaps something in the UEFI API already tackles that?

sunnywang-arm, 2021-04-12:
Just sent a patch to the edk2 mailing list for adding a setup option with the default ConnectAll setting.  
https://edk2.groups.io/g/devel/topic/patch_1_1/82033650?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,82033650

RafaelMarinheiro, 2021-04-12:
I tried the patch and it worked like a charm (ignoring mail encoding issues). There are some minor issues in the text (missing whitespaces/typos) but that was already pointed out in the review.

Thank you very much for this!

0n0w1c, 2021-04-12:
Thanks to Rafael for building and sharing, I have tried the patched RPi_EFI.fd and it does fix the Windows ARM booting issue.
Thank you @sunnywang-arm for the patch, it works great!

sunnywang-arm, 2021-04-13:
Thanks for your quick follow-up/testing, @RafaelMarinheiro and @0n0w1c   

As for the mail encoding issues, I may find the root cause and fixed it in v2 (https://edk2.groups.io/g/devel/message/74014). If you still see the mail encoding issues, feel free to let me know.

sunnywang-arm, 2021-04-16:
Close this issue because @ardbiesheuvel already pushed the fix and the fix was already verified by @RafaelMarinheiro and @0n0w1c. Thanks, guys.

closed 2021-04-16 by sunnywang-arm