pftf/RPi4 #145: total_mem not working as expected

RgA59, 2021-04-21:
I'm sorry if I'm not following established protocol but I'm a total noob at Github.

Raspberry Pi 4B v1.4 8GB in UEFI mode (WoR), I disable the 3GB memory limit and then I add **total_mem=4096** to config.txt.
After a reboot the OS still sees the full 8GB as usable.

Am I doing it wrong, is this a limitation, by design or a bug?

mariobalanica, 2021-04-21:
`total_mem` only works on Linux as far as I know.
For Windows, you need to set the `truncatememory` BCD setting: https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/boot-parameters-to-manipulate-memory

jlinton, 2021-04-21:
The config.txt option may not even work in Linux. The standard Linux way to do this is 'mem=4G' on the kernel command line in grub.

RgA59, 2021-04-21:
> `total_mem` only works on Linux as far as I know.
> For Windows, you need to set the `truncatememory` BCD setting: https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/boot-parameters-to-manipulate-memory

Thanks for your response and wow! coming from the project's boss none the less. Congrats! you're doing a great job, please keep it up!
I just got my first Pi4 recently (specifically to try your project), being a total noob at all this including Linux, I find it odd that other Raspian command lines in config.txt lines work but total_mem doesn't.
Now..., the reason I'm trying to limit memory is because in GPT mode with 8GB usable Windows crashes constantly. If I flash in MBR mode, no crashes but Windows doesn't see more than 4GB. I'd like to find a sweet spot where I can use GPT with more than 4GB without crashing.
I'm in the process of flashing today's new build so I'll use the  "removememory" command as recommended in the link you provided and see if that works.
Any other ideas or advise you might have would be greatly appreciated. Thanks in advance. Cheers!

RgA59, 2021-04-21:
> The config.txt option may not even work in Linux. The standard Linux way to do this is 'mem=4G' on the kernel command line in grub.

It works in Raspbian non-UEFI both 32 and 64bit

mariobalanica, 2021-04-22:
Can you open an issue here: https://www.worproject.ml/bugtracker with the latest log file from the imager?

RgA59, 2021-04-22:
> Can you open an issue here: https://www.worproject.ml/bugtracker with the latest log file from the imager?

I'd be glad to but I'm a bit confused.
If in fact total_mem doesn't apply to WoR or Windows as you mentioned, then there's no issue, is there?
On the other hand, I've tried bcdedit "removememory" and it doesn't work either. I'm stuck at Pi 3GB limit, 4GB in MBR mode with limit disable or the full 8GB usable in GPT mode but with constant GSODs.

RgA59, 2021-04-23:
https://www.worproject.ml/bugtracker/view.php?id=140#c239

closed 2021-04-23 by RgA59