pftf/RPi4 #146: Cannot boot Windows from eMMC on CM4

AnaJade, 2021-04-22:
I noticed that the booting process cannot complete when Windows is installed on the eMMC of a CM4.
The boot seems to freeze once the UEFI configuration period is complete. I can modify settings in the UEFI without any problems, but I can't finish the boot after that. The screen stays stuck on the Raspberry Pi logo, and the Windows spinning dots never start.
 
I used a CM4 with a 16GB eMMC and 4GB of RAM.

korhojoa, 2021-04-27:
I have the same experience with the UUGear RasPiKey. http://www.uugear.com/product/raspikey-plug-and-play-emmc-module-for-raspberry-pi/
32GB eMMC and 8GB of RAM.

robo523, 2021-05-02:
same problem here.  CM4 with 32 eMMC and 4GB ram.  I tried the fix noted in the 1.23 release and I still have the same issue.  I know the windows ISO im using works on a rpi4b.  The chipset should be the same, so I don't know if this is a UEFI issue or a windows issue using the eMMC memory.|
system locks up before the windows dots go in the circle.  I can see the dots on a normal rpi4b, nothing on the CM4.

jlinton, 2021-05-19:
Part of the problem might be the firmware images provided here aren't really compatible with the cm4 without the PCIe/XHCI switch, which is hung up in various review processes. If someone wants to debug this a bit either remove the XHCI nodes from the acpi tables, or cherry pick the xhci/pci menu switch from one of my trees and see if that works.

jlinton, 2021-09-01:
I think this might be fixed in the .30 version now available here. (Thats assuming windows is trying to probe a XHCI adapter that only exists on the RPI4b and not the cm4).
Closing, reopen if the latest build is still failing.

closed 2021-09-01 by jlinton

halentech, 2021-11-04:
Version 1.32 still not booting on NEC CM4 module. Please reopen.

Koteyk0o, 2022-01-25:
I confirm. CM4, still stuck on UEFI screen. At this point USB stops working. No keyboard, no USB SSD, nothing works.

mnoorianp, 2022-02-10:
same problem here. CM4 with 32 eMMC and 4GB ram.

etienne51, 2024-03-05:
Also tried it on my CM4 8GB on microSD. Stuck on UEFI for a while, then starts booting and throws BSOD a few sec after and restarts. It's my uConsole with an external display (internal not showing anything).