pftf/RPi4 #148: ALTERNATE GUIDE – RUNNING WINDOWS 10 ON THE PI 4

RgA59, 2021-05-10:
The guide at https://rpi4-uefi.dev/ needs updating, it dates back to July 2020.
Please update it. Thanks in advance.

pbatard, 2021-05-13:
Can you clarify what you think needs updating?

I suppose you might be suggesting to use a more recent version of Windows 10 but the problem is we do want to work with __release__ versions (not insider ones, since these cannot be considered stable), and, as long as we have not seen how the release version of 21H1 fares in terms of Pi 4 support, version 20H1 (19041.330) is still the only version we can recommend for compatibility.

So before we can look at updating the guide, we need to understand what you think is outdated there.

RgA59, 2021-05-13:
Hi. Thanks for replying.
Yes, that's what I meant but you made a very good point, I understand your concerns so disregard.
FYI: 19043. 985 it's my daily driver, very stable and doesn't brake anything. Dev 21376 is also stable and working as expected, I even figured how to get BT working. You can check that at Discord #wor-support.
Anyway, you guys are doing a great job, please keep it up!
Cheers!

pbatard, 2021-05-13:
Okay. We will obviously check 21H1 when it's released and update the guide as needed.

I will close this issue then.

closed 2021-05-13 by pbatard