pftf/RPi4 #149: CPU Info Linux

nifares2007, 2021-05-16:
Hello,
In linux no clock rate is displayed, CPU model is not displayed either, I would be happy if you would fix that.

jlinton, 2021-05-18:
Where are you looking in linux?
[root@rpi4-f31 ~]# lscpu
Architecture:                    aarch64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
CPU(s):                          4
On-line CPU(s) list:             0-3
Thread(s) per core:              1
Core(s) per socket:              4
Socket(s):                       1
NUMA node(s):                    1
Vendor ID:                       ARM
Model:                           3
Model name:                      Cortex-A72
Stepping:                        r0p3
Frequency boost:                 disabled
CPU max MHz:                     1500.0000
CPU min MHz:                     600.0000
...

The CPU freq there is the ranges, for the current frequency you need to use cpupower 

[root@rpi4-f31 btrfs-progs.mainline]# cpupower frequency-info
analyzing CPU 0:
  driver: cppc_cpufreq
  CPUs which run at the same hardware frequency: 0
  CPUs which need to have their frequency coordinated by software: 0
  maximum transition latency:  Cannot determine or is not supported.
  hardware limits: 600 MHz - 1.50 GHz
  available cpufreq governors: conservative ondemand userspace powersave performance schedutil
  current policy: frequency should be within 600 MHz and 1.50 GHz.
                  The governor "performance" may decide which speed to use
                  within this range.
  current CPU frequency: 1.50 GHz (asserted by call to hardware)

but that functionality requires out of tree patches for tfa at the moment.

nifares2007, 2021-05-19:
I don't want to use any other sopftware to get information, the Uefi bios should speak to the PC standard that the bios properly transmits the information, with phpsysinfo 3 and other php where I can read cpu it shows me no cpu clock and how can and it also does not show the cpu model as Broadcom BCM2711

root@home:~# lscpu
Architecture:                    aarch64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
CPU(s):                          4
On-line CPU(s) list:             0-3
Thread(s) per core:              1
Core(s) per socket:              4
Socket(s):                       1
NUMA node(s):                    1
Vendor ID:                       ARM
Model:                           3
Model name:                      Cortex-A72
Stepping:                        r0p3
BogoMIPS:                        108.00
NUMA node0 CPU(s):               0-3
Vulnerability Itlb multihit:     Not affected
Vulnerability L1tf:              Not affected
Vulnerability Mds:               Not affected
Vulnerability Meltdown:          Not affected
Vulnerability Spec store bypass: Vulnerable
Vulnerability Spectre v1:        Mitigation; __user pointer sanitization
Vulnerability Spectre v2:        Vulnerable
Vulnerability Srbds:             Not affected
Vulnerability Tsx async abort:   Not affected
Flags:                           fp asimd evtstrm crc32 cpuid

nifares2007, 2021-05-19:
root@home:~# cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

nifares2007, 2021-05-19:
unfortunately i don't speak english i accidentally closed it how can i open it again

jlinton, 2021-05-19:
The CPU model is shown as bcm2711 in the smbios data (use dmidecode), of which this is firmware provided:

Handle 0x0007, DMI type 4, 48 bytes
Processor Information
        Socket Designation: Socket
        Type: Central Processor
        Family: ARM
        Manufacturer: Broadcom
        ID: 83 D0 0F 41 00 00 00 00
        Signature: Implementer 0x41, Variant 0x0, Architecture 15, Part 0xd08, Revision 3
        Version: BCM2711 (ARM Cortex-A72)
        Voltage: 5.0 V 3.3 V 2.9 V
        Max Speed: 1700 MHz
        Current Speed: 1700 MHz
        Status: Populated, Enabled
        L1 Cache Handle: 0x0005
        L2 Cache Handle: 0x0006
        Core Count: 4
        Core Enabled: 4
        Thread Count: 4
        Characteristics:
                64-bit capable
                Multi-Core
                Execute Protection
                Enhanced Virtualization

/proc/cpuinfo isn't a firmware function, that is 100% linux kernel. There have been dozens of posted patches (including one sent by me) to LAML to "fix" this with additional fields, and they have all been soundly rejected because the maintainers aren't interested in providing information similar to what x86  provides in /proc/cpuinfo. They believe it is a deprecated interface. So, they don't want to attempt to maintain MIDR to vendor tables, or do smbios lookups in the kernel. x86 machines provide the "Intel(R) Core(TM) i3-9100 CPU @ 3.60GHz" strings you see in /proc/cpuinfo via the cpuid instruction on the core, arm cores don't have similar human readable funcitonality instead depending on firmware provided tables, which I believe is being provided by this firmware.  So, if you don't want to rely on a utility to decode this information you can decode it from the files in /sys/firmware/dmi/entries/ or some of the areas which are 1st party sources in the kernel. Just understand its not going to be as nice as lscpu (which displays the bios model in more recent versions)

[root@rpi4-f31 util-linux]# ./lscpu
Architecture:           aarch64
  CPU op-mode(s):       32-bit, 64-bit
  Byte Order:           Little Endian
CPU(s):                 4
  On-line CPU(s) list:  0-3
Vendor ID:              ARM
  BIOS Vendor ID:       Broadcom
  Model name:           Cortex-A72
    BIOS Model name:    BCM2711 (ARM Cortex-A72)
    Model:              3
    Thread(s) per core: 1
    Core(s) per socket: 4
    Socket(s):          1
    Stepping:           r0p3
    Frequency boost:    disabled
    CPU max MHz:        1500.0000
    CPU min MHz:        600.0000
    BogoMIPS:           108.00
    Flags:              fp asimd evtstrm crc32 cpuid
Caches (sum of all):
  L1d:                  128 KiB (4 instances)
  L1i:                  192 KiB (4 instances)
  L2:                   1 MiB (1 instance)
NUMA:
  NUMA node(s):         1
  NUMA node0 CPU(s):    0-3

nifares2007, 2021-05-19:
Unfortunately, 99.99 percent of the programs work with cpuinfo and not with lscpu, I think you program or are not involved in this uefi bios, if you already have an x86 uefi bios then you should program it correctly and not this client-side kernel , or it is not technologically possible because it does not work with arm [risc]

jlinton, 2021-05-20:
This project doesn't provide Linux kernels, and as I understand it, your complaint is with /proc/cpuinfo's lack of meaningful information. I would suggest that you open a bug against mainline Linux here: https://bugzilla.kernel.org/

I believe you will have more luck there as I don't think this is a firmware issue. So I'm going to close this one. 

Thanks,

closed 2021-05-20 by jlinton

nifares2007, 2021-05-20:
this is not a kernel problem but of the bios the uefi bios gives less information than the stanrd submarine from raspberry pi i think is just an optimization of this project