pftf/RPi4 #152: CM4 unable to detect its USB2.0 hub, thus unable to boot into shell/setup

adam900710, 2021-06-04:
The involved release is v1.27.

The board is CM4 with 8G RAM, 32G eMMC, with wifi/bt.

I changed the rpi bootloader to boot from BCM-USB-MSD first, then eMMC.

Although it can boot into the RPI icon screen, waiting for keyboard input, the keyboard plugged has no response at all, thus no way to interrupt.

It looks like the DWC2 usb controller is either not detected (wrong device tree?) nor no driver for it?

The problem is the same when using Manjaro linux-rpi-mainline kernel, which can't see the DWC2 controller at all.

adam900710, 2021-06-04:
DWC2 dt overlay is not included, thus it can't enable the only USB hub in CM4, thus no way to use USB keyboard.

Please at least that overlay for CM4 support.

jwillikers, 2021-06-05:
You can connect over UART to interact with the UEFI console on the CM4. Although,with the IO board, I imagine booting something over USB won't work until the bus is enabled by the UEFI. The UEFI works great when flashed to the on-board eMMC, but for my use case it doesn't really help if I can't boot an ISO over USB.

adam900710, 2021-06-05:
Yep, exactly.
Without the dwc2 device tree overlay added, no usb hub at all, thus no way to boot from any USB device.

tetofonta, 2021-07-24:
I've the same problem. Everything is fine when running from emmc and the usb is initialized and the keyboard works.
The issue comes when booting from usb (eeprom boot sequence configured as BCM-USB-MSD -> emmc -> network): the usb key is found and the pi boots from it, but after everything is loaded nothing is recognized and thus the keyboard don't work.
I've inserted `dtoverlay=dwc2,dr_mode=host` in config.txt and the relative overlay hoping to enable the usb bus but it did not work.

I'm using version 1.28

andreiw, 2022-05-04:
CM4 IO board + CM4... USB2 working in edk2. Booting firmware from USB2 as well. Here is what my config.txt looks like.

I've tried adding dtoverlay=dwc2,dr_mode=host as well.

arm_64bit=1
enable_uart=1
uart_2ndstage=1
enable_gic=1
armstub=RPI_EFI.fd
disable_commandline_tags=1
disable_overscan=1
device_tree_address=0x1f0000
device_tree_end=0x200000
dtoverlay=miniuart-bt
devicetree=bcm2711-rpi-cm4.dtb

andreiw, 2022-05-04:
And yeah no idea why rpiboot doesn't work for me, so I can't really boot from emmc...