pftf/RPi4 #153: Please provide option to boot 8Gb

Sc0th, 2021-06-07:
Would be super helpful if there was a method to allow 8Gb on boot rather than 3Gb limit, running a cluster of headless Pi's running ESXi ARM fling means it it incredibly time consuming to consider a UEFI upgrade, as they are all booting from tftp server being able to remove the limit on the images would be a godsend!

samerhaj, 2021-07-07:
You can customize the UEFI configuration settings to disable the 3GB RAM limit, and save that (which saves the settings in the image itself). Then use that new image for the PXE boot (copy it from SD to the TFTP server).

You can also script the change on multiple systems by booting to the UEFI Shell and running a startup.nsh file from any media. Examples of how to do this are documented here: https://github.com/tianocore/edk2-platforms/tree/master/Platform/RaspberryPi/RPi4#configuration-settings

Sc0th, 2021-07-07:
Sadly it's not a simple as that (unless I am missing a trick) as whilst that does work, the act of saving the changes embeds the MAC of the current host into the image, so whilst this can be copied away to a TFTP server for remote boot, it can _only_ be used to boot the Pi on which the configuration change was made.

Ergo this needs to be done for each Pi and the TFTP server needs to hold a separate image for each host, which is not ideal. However... there is no requirement to have the MAC in the UEFI image to actually boot the PI vi TFTP so in an ideal world we could place a single image on the TFTP server, set it to utilise 8Gb and have all hosts boot from the same image. As it stands upgrading the UEFI on a headless cluster of Pi's is at best a PITA.

I do apologise (and would be pleased!) if I am missing something here!

Sc0th, 2021-07-08:
This what I refer to:

`[INF] infra01:~# arp -a | grep pinode01
pinode01.<mydomain> (192.168.60.205) at dc:a6:32:c5:df:c5 [ether] on ens192
[INF] infra01:~# xxd /appdata01/PROD/pikick/data/tftpboot/738ef2ab/RPI_EFI.fd | grep -B1 A.6.3.2.C.5.D.F
001d1450: 2000 2800 4d00 4100 4300 3a00 4400 4300   .(.M.A.C.:.D.C.
001d1460: 4100 3600 3300 3200 4300 3500 4400 4600  A.6.3.2.C.5.D.F.
--
001d1590: 2000 2800 4d00 4100 4300 3a00 4400 4300   .(.M.A.C.:.D.C.
001d15a0: 4100 3600 3300 3200 4300 3500 4400 4600  A.6.3.2.C.5.D.F.
[INF] infra01:~#
`

samerhaj, 2021-07-17:
This request is also reported by others in the comments on this article: https://williamlam.com/2020/11/stateless-esxi-arm-with-raspberry-pi.html

paulwratt, 2022-02-18:
Is there some way to (optionally) preload default settings, in the same way that this project overrides the F9 defaults.

If so that would also relate to the need to pre-define GPIO settings (per machine or per instance) mentioned elsewhere.

pbatard, 2022-03-07:
I think the best way to work around the 3 GB limit is to compile your own firmware, as I don't see us providing multiple versions of the UEFI firmware with and without the 3 GB default. Please remember that the firmware is targeted at satisfying general rather than specific usage, and therefore we cannot ditch the 3 GB limit that easily as some environments need it. So if you need something more specific, you should recompile your own firmware with `--pcd gRaspberryPiTokenSpaceGuid.PcdRamLimitTo3GB=0` or edit the default environment variables.

As such, I will close this issue.

closed 2022-03-07 by pbatard