pftf/RPi4 #154: The default UART settings in config.txt do not work properly under Windows IOT

sunnywang-arm, 2021-06-09:
According to https://www.raspberrypi.org/documentation/configuration/uart.md, the current UART setting in config.txt doesn't make sense, and I also found it doesn't work under Windows IOT.  
- With the current default setting ("enable_uart=1" and "dtoverlay=miniuart-bt"),  I can't get the serial message by using Putty, and it doesn't make sense to switches the Bluetooth function to use the mini UART ("dtoverlay=miniuart-bt") when we're using it as Primary UART ("enable_uart=1"). 
- With "enable_uart=1" only setting (Remove "dtoverlay=miniuart-bt"), I can only get garbage message. After adding "core_freq=250", it works properly. It looks like the core clock hasn't really been fixed to 250MHz as what being mentioned in uart.md.

sunnywang-arm, 2021-06-09:
I think we should change the config,txt to the following for making both UART and Bluetooth work properly: (Remove dtoverlay=miniuart-bt and add core_freq=250). 
```
arm_64bit=1
uart_2ndstage=1
enable_gic=1
armstub=RPI_EFI.fd
disable_commandline_tags=1
disable_overscan=1
device_tree_address=0x1f0000
device_tree_end=0x200000
enable_uart=1
core_freq=250
```
@pbatard @samerhaj @andreiw @ardbiesheuvel @jlinton @jaredmcneill @PhatFree what do you guys think? If no objection, I will update config.txt.

Moreover, I will report this issue to [WOR project](https://www.worproject.ml/bugtracker/my_view_page.php) as well. If this issue also needs to be reported to somewhere else, feel free to let me know.

sunnywang-arm, 2021-06-09:
Add a note here for others who would like to change the primary UART to PL011
- For using PL011 as the primary UART, we should have "dtoverlay=miniuart-bt" and "core_freq=250" and remove "enable_uart=1"
- For using Mini UART as the primary UART,  we should have "enable_uart=1" and "core_freq=250" and remove dtoverlay=miniuart-bt.

ardbiesheuvel, 2021-06-09:
> Add a note here for others who would like to change the primary UART to PL011
>
> * For using PL011 as the primary UART, we should have "dtoverlay=miniuart-bt" and "core_freq=250" and remove [...]

Doesn't setting core_freq= interfere with the system's ability to manage the clocks based on system load? The UART is a debug tool, so it should not be enabled by default IMHO.

sunnywang-arm, 2021-06-09:
> 
> Doesn't setting core_freq= interfere with the system's ability to manage the clocks based on system load? The UART is a debug tool, so it should not be enabled by default IMHO.

Good point! You're right. @ardbiesheuvel 
In this case, we may just need to remove "dtoverlay=miniuart-bt" and update the Readme.md. What do you think?

pbatard, 2021-06-09:
I don't think we want to do what you suggest, as we use PL011 as default UART to avoid losing serial on Linux as the system throttles, and, unlike what is the case for Windows, we expect to have a significant set of Linux installations and usage being carried out exclusively through serial, in which case we must have stable serial (without crippling the core freq). That's pretty much the reason why the current defaults for `config.txt` are what they are.

So my take on this is that we can add a note in the Readme to let Windows users, who want to use serial/putty, know how they should alter their `config.txt`. But I don't think we want to change the current defaults, as it's likely to inconvenience a lot more people on Linux than it'll benefit Windows users.

mariobalanica, 2021-06-10:
This is actually an issue in the Windows drivers: https://github.com/raspberrypi/windows-drivers/issues/33

sunnywang-arm, 2021-06-10:
Thanks, @pbatard and @mariobalanica 

Since this is a known Windows driver issue and Linux OSes are working fine with the current config.txt, so I think we can close this issue now. 

I will spare time to check why Linux OSes can still work fine with the conflict between "enable_uart=1" and "dtoverlay=miniuart-bt" described in https://www.raspberrypi.org/documentation/configuration/uart.md. My guess now is that "dtoverlay=miniuart-bt" will override "enable_uart" to 0 because "dtoverlay=miniuart-bt" is the last line in config.txt.

closed 2021-06-10 by sunnywang-arm