pftf/RPi4 #157: Unable to from USB xhci_hcd deregistered - Kernel 5.12.0+

xkotj, 2021-06-16:
Hello,
I am having a issue with my RPI 4B 8GB. 
EFI on SD CARD and system on SSD connected via USB.

I have been using Fedora Server from version 33. Now I have latest version 34 and it was working as expected.
Configured devicetree in EFI, 8GB enabled. Kernel was 5.11.x. System booted and I could see /dev/i2c-1 which I used a lot.

Now after upgrading kernel to 5.12.x and having configured  device tree in EFI it gets stuck on deregistering all USBs.
I am unable to boot from my external SSD nor use keyboard. Does anyone have same problem as me with latest kernels?
Is there any workaround for this?

I tried to boot with ACPI configured in EFI and I was able to get to the system, but I am not able to detect any i2c bus which I need.

Could anyone please point me what can be done?
thanks.

[image]

xkotj, 2021-07-02:
Also one more thing. When I install newer kernel I get warning on screen.
power/level is deprecated; use power/control instead.
Could it be related?

wzzrd, 2021-07-02:
Same issue. Forcing me to stay on 5.11