pftf/RPi4 #158: systemctl reboot --firmware-setup doesn't work

jlinton, 2021-06-23:
The subject is somewhat self-explanatory, although come to think about it, I'm not sure I've ever tried this on a arm64 before. The grub setup option does work, so its likely something to do with the uefi variables misbehaving again.

jlinton, 2021-06-23:
System is of course fedora rawhide with a hand build 5.13rc6.

samerhaj, 2021-06-23:
For this to work, the OS writes a NV UEFI variable called OsIndications (and the UEFI defined global variables GUID), with the value setting the bit EFI_OS_INDICATIONS_BOOT_TO_FW_UI. The variable needs to persist across reboots, so the UEFI FW can see it on next reboot, and act accordingly. The code to do so is in the main BDS boot path that we have on the RPi (https://github.com/tianocore/edk2/blob/master/MdeModulePkg/Universal/BdsDxe/BdsEntry.c#L561)

So this essentially becomes an issue of persisting UEFI NV variables, which we know (https://github.com/pftf/RPi4/issues/6)

jlinton, 2021-12-03:
Definitely a variable store bug, as it works now. Closing as a dupe of https://github.com/pftf/RPi4/issues/6

closed 2021-12-03 by jlinton