pftf/RPi4 #159: Booting from nvme devices on CM4

potatoigit, 2021-06-27:
Could it CM4 boot from nvme devices via the tianocore loader?

jlinton, 2021-06-28:
Yes, but at the moment there isn't any linux kernel support for it. The patches are very slowly working their way through the system as we are trying to adhere to a new arm standard https://developer.arm.com/documentation/den0115/latest which avoids needing custom PCIe drivers for every machine that isn't 100% compliant with the PCIe specification.

There have been a couple unexpected speed bumps unrelated to the technical difficulties of doing this. The patches have been on various mailing lists for a few months now. for example some of the LKML discussion https://lore.kernel.org/linux-pci/20210620002656.GL1002214@nvidia.com/T/

potatoigit, 2021-06-28:
Thanks a lot and waiting your good news :D

closed 2021-06-28 by potatoigit