pftf/RPi4 #160: Win 10 ignoring overclock setting in UEFI

stanton2, 2021-07-07:
I'm running a RPi 4b with 8GHz RAM.
I've been told to overclock using Raspberry OS first, via boot/config/txt.  Then insert the Win10 sd card and configure the UEFI to show the same overclock.
The overclock to 2 GHz using Raspberry OS works perfectly and the "watch" command displays cpu speeds varying between 600 MHz and 2 GHz with load.
However, the saved cpu configuration of 2 GHz  in the UEFI is not reflected in the Task Manager, which continues to display 1.5 GHz as the base speed
This is my first attempt on Github, so trust that the information provided is meaningful.
Thank you in advance for any assistance provided.
Adrian

samerhaj, 2021-07-07:
This is duplicate of #39

samerhaj, 2021-07-17:
Closing as duplicate

closed 2021-07-17 by samerhaj