pftf/RPi4 #161: Manually set resolution is overridden when there is no display connected

BenNottelling, 2021-07-09:
Title basically says it all. If I manually set a resolution such as 1280x720 it gets overridden if the pi is booted without a display connected.
This means if I have a pi setup to be a headless system to remote into and it has to reboot for whatever reason, I'm locked into an extremely low resolution such as 640x480 and with windows 10 not having display drivers, I have no way of changing it.
The only fix is to physically connect to a display and then reboot the pi. Then the UEFI specified resolution is used.

TheMindVirus, 2021-07-10:
[image]

BenNottelling, 2021-07-11:
@TheMindVirus Thanks for giving a visual example of what I mean.
I was looking at your VC4WDDM2 repository and saw the Pi4DXGK-dev branch.
Noticed inside ```Pi4BDD.CXX``` it seems you've hardcoded 1920x1080. I think if I'm understanding it correctly you're working on a WDDM display and renderer driver? I would assume once finished that'll actually let me change the resolution in windows. Until then, can I hardcode 1280x720 and compile it myself? Figured I'd check before I brick another windows install on my pi

Thanks

TheMindVirus, 2021-07-11:
I wouldn't normally hardcode 1920x1080 but it seems necessary as the process to switch resolution has been somewhat temperamental. I even added the resolution switcher back in on the dev branch as you saw, but Windows complains that settings fail to save and the actual change almost never persists reliably.