pftf/RPi4 #162: UEFI Errors out when no screen is connected

AmirDahan, 2021-07-12:
If you planned to boot up your pi without a screen connected you will need to add
a custom resolution
or hdmi_force_hotplug=1
There is an issue where the uefi wont boot if it doesn't detect a screen

A fix for this would be to make uefi check if no screen is detected and continue booting with a virtual resolution of 640x480

Wyzard256, 2021-08-28:
Headless boot is working for me in firmware 1.29, using the default resolution setting ("native resolution") and without that hotplug line in config.txt.  If I boot headless and then connect a monitor afterward, I get no signal (the monitor immediately goes into sleep mode, even if I press a key to wakeup the Linux console), but the OS still works and can be accessed via network.

(I don't know if this means the bug is fixed, or if I'm just misunderstanding how to reproduce it.)

For reference, my config.txt says:

    arm_64bit=1
    enable_uart=1
    uart_2ndstage=1
    enable_gic=1
    armstub=RPI_EFI.fd
    disable_commandline_tags=1
    disable_overscan=1
    device_tree_address=0x1f0000
    device_tree_end=0x200000
    dtoverlay=miniuart-bt

paulwratt, 2022-02-18:
`hdmi_force_hotplug=1` should allow later HDMI connection and activation.

Not having HDMI become active later implies _no framebauffer_ allocation at boot time.

dimpase, 2022-12-14:
For the current firmware, 1.33, I also had to add `hdmi_force_hotplug=1` to let the box run in headless mode.
I recall setting monitor resolution to 1080 (although it had not affected its working, as far as was able to see).

Frankly it's the last thing I expected from firmware, to force a monitor present by default, so in the end I had to run around with an old  bulky 22" monitor and a keyboard  to diagnose and fix. :smile:

closed 2023-03-27 by AmirDahan