pftf/RPi4 #164: ACS 3.0 - SetTime_Func failures

samerhaj, 2021-07-16:
ACS 3.0 on 1.28 FW on RPi400, seen SetTime_Func failures. Full log below in comment

Changing the time (and preserving across reboots) is expected to fail on the RPi since there is no RTC hardware on the RPi (https://github.com/pftf/RPi4/issues/7 ) and the UEFI time services are implemented using an emulated VirtualTime driver without support for preserving time across reboots. 

However, looking at these RPi400 failures, there are multiple issues with failure in changing the date, timezone, and daylight savings. These should not fail , as they simply store UEFI NV variables.

samerhaj, 2021-07-16:
```
------------------------------------------------------------
------------------------------------------------------------
Arm ACS Version: v3.0
SetTime_Func
Revision 0x00010000
Test Entry Point GUID: 603B46BE-7E14-408A-93D7-DD9DEC732968
Test Support Library GUIDs: 
  1F9C2AE7-F147-4D19-A5E8-255AD005EB3E
------------------------------------------------------------
UEFI 2.6
Test Configuration #0
------------------------------------------------------------
Interface Test for SetTime
------------------------------------------------------------
Logfile: "\SCT_RUN\Overall\Summary.log"
Test Started: 06/16/21  04:35p
------------------------------------------------------------
RT.SetTime - Change year -- FAILURE
29151AE4-7A5E-42D9-84F8-E9C56787B7E8
/home/edhcha01/ACSv3.0REL/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.2+gitAUTOINC+ed8a7477d4-r0/SctPkg/TestCase/UEFI/EFI/RuntimeServices/TimeServices/BlackBoxTest/TimeServicesBBTestFunction.c:313:Status - Invalid Parameter, TPL - 4, Year - 2021

------------------------------------------------------------
      *********RECOVERY*********
------------------------------------------------------------
RT.SetTime - Verify year after change -- FAILURE
3B96A20C-2B1F-44EA-BAA9-F96FEE131D05
/home/edhcha01/ACSv3.0REL/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.2+gitAUTOINC+ed8a7477d4-r0/SctPkg/TestCase/UEFI/EFI/RuntimeServices/TimeServices/BlackBoxTest/TimeServicesBBTestFunction.c:385:Status - Success, TPL - 4

RT.SetTime - Change month -- FAILURE
4E123824-8636-4426-81E6-1675628CDE69
/home/edhcha01/ACSv3.0REL/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.2+gitAUTOINC+ed8a7477d4-r0/SctPkg/TestCase/UEFI/EFI/RuntimeServices/TimeServices/BlackBoxTest/TimeServicesBBTestFunction.c:425:Status - Invalid Parameter, TPL - 4

------------------------------------------------------------
      *********RECOVERY*********
------------------------------------------------------------
RT.SetTime - Verify month after change -- FAILURE
2D5CDBE5-1055-4EF6-8E90-0C993F93F698
/home/edhcha01/ACSv3.0REL/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.2+gitAUTOINC+ed8a7477d4-r0/SctPkg/TestCase/UEFI/EFI/RuntimeServices/TimeServices/BlackBoxTest/TimeServicesBBTestFunction.c:497:Status - Success, TPL - 4

RT.SetTime - Change daylight -- FAILURE
7AF90CE7-1FED-4101-82DC-CC634CDF204E
/home/edhcha01/ACSv3.0REL/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.2+gitAUTOINC+ed8a7477d4-r0/SctPkg/TestCase/UEFI/EFI/RuntimeServices/TimeServices/BlackBoxTest/TimeServicesBBTestFunction.c:536:Status - Invalid Parameter, TPL - 4

------------------------------------------------------------
      *********RECOVERY*********
------------------------------------------------------------
RT.SetTime - Verify daylight after change -- PASS
B39BC904-55E7-4B9B-B4D8-274ADD71D625
/home/edhcha01/ACSv3.0REL/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.2+gitAUTOINC+ed8a7477d4-r0/SctPkg/TestCase/UEFI/EFI/RuntimeServices/TimeServices/BlackBoxTest/TimeServicesBBTestFunction.c:606:Status - Success, TPL - 4

RT.SetTime - Change time zone -- FAILURE
CDBBDA04-4F7C-4BA5-8BCF-C050E5A976C7
/home/edhcha01/ACSv3.0REL/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.2+gitAUTOINC+ed8a7477d4-r0/SctPkg/TestCase/UEFI/EFI/RuntimeServices/TimeServices/BlackBoxTest/TimeServicesBBTestFunction.c:642:Status - Invalid Parameter, TPL - 4

------------------------------------------------------------
      *********RECOVERY*********
------------------------------------------------------------
RT.SetTime - Verify time zone after change -- PASS
EA99DEC5-B879-4C8D-BFD1-F63FE75899BF
/home/edhcha01/ACSv3.0REL/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.2+gitAUTOINC+ed8a7477d4-r0/SctPkg/TestCase/UEFI/EFI/RuntimeServices/TimeServices/BlackBoxTest/TimeServicesBBTestFunction.c:714:Status - Success, TPL - 4

RT.SetTime - Restore time -- FAILURE
6A8CAA83-B9DA-46C7-98F6-D4969DABDAA0
/home/edhcha01/ACSv3.0REL/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.2+gitAUTOINC+ed8a7477d4-r0/SctPkg/TestCase/UEFI/EFI/RuntimeServices/TimeServices/BlackBoxTest/TimeServicesBBTestFunction.c:734:Status - Invalid Parameter, TPL - 4

RT.SetTime - Change year -- FAILURE
75E988EE-EC78-4190-9A09-B1315C2025A5
/home/edhcha01/ACSv3.0REL/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.2+gitAUTOINC+ed8a7477d4-r0/SctPkg/TestCase/UEFI/EFI/RuntimeServices/TimeServices/BlackBoxTest/TimeServicesBBTestFunction.c:313:Status - Invalid Parameter, TPL - 8, Year - 2021

[... 27 lines cut ...]
/home/edhcha01/ACSv3.0REL/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.2+gitAUTOINC+ed8a7477d4-r0/SctPkg/TestCase/UEFI/EFI/RuntimeServices/TimeServices/BlackBoxTest/TimeServicesBBTestFunction.c:606:Status - Success, TPL - 8

RT.SetTime - Change time zone -- FAILURE
F749B4F1-537D-4DDF-8545-C0A41993CEE4
/home/edhcha01/ACSv3.0REL/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.2+gitAUTOINC+ed8a7477d4-r0/SctPkg/TestCase/UEFI/EFI/RuntimeServices/TimeServices/BlackBoxTest/TimeServicesBBTestFunction.c:642:Status - Invalid Parameter, TPL - 8

------------------------------------------------------------
      *********RECOVERY*********
------------------------------------------------------------
RT.SetTime - Verify time zone after change -- PASS
D9C645B9-52DE-415C-ABDC-7226CE6A30B1
[... cut ...]
RT.SetTime - Restore time -- FAILURE
6A8CAA83-B9DA-46C7-98F6-D4969DABDAA0
/home/edhcha01/ACSv3.0REL/arm-enterprise-acs/luv/build/tmp/work/aarch64-oe-linux/sbbr/v1.2+gitAUTOINC+ed8a7477d4-r0/SctPkg/TestCase/UEFI/EFI/RuntimeServices/TimeServices/BlackBoxTest/TimeServicesBBTestFunction.c:734:Status - Invalid Parameter, TPL - 8

Returned Status Code: Success

SetTime_Func: [FAILED]
  Passes........... 4
  Warnings......... 0
  Errors........... 14
------------------------------------------------------------
```

sunnywang-arm, 2021-07-23:
This is an FW issue. 
I made and sent a fix to the edk2 mailing list - https://edk2.groups.io/g/devel/topic/patch_1_1/84397263.

sunnywang-arm, 2021-07-23:
By the way, this FW issue also caused problem with shell time and date commands. 
```
Shell> time 9:51:30
time: Invalid argument - '9:51:30'
Shell> date
06/16/2021
Shell> date 07/22/2021
date: Invalid argument - '07/22/2021'
```

sunnywang-arm, 2021-08-02:
The fix got merged, so close it. Thanks, @ardbiesheuvel and @pbatard 
https://github.com/tianocore/edk2/commit/8e6bb64fe43e5be7bf3becfbc5dbcdc90ab6425c

closed 2021-08-02 by sunnywang-arm