pftf/RPi4 #167: Boot hangs for pcie related assertion only with debug binary

tetofonta, 2021-07-28:
I'm trying to make this software work on a CM4 1GB booting from msd (`BCM-USB-MSD`).
The usb controller does not work, so i tried to fire up the debug version of v1.28 to catch any errors during boot and i can't reach the dwc initialization part because of this error loading `PciHostBridgeDxe.efi`

```
add-symbol-file /home/runner/work/RPi4/RPi4/Build/RPi4/DEBUG_GCC5/AARCH64/MdeModulePkg/Bus/Pci/PciHostBridgeDxe/PciHostBridgeDxe/DEBUG/PciHostBridgeDxe.dll 0x36C97000
Loading driver at 0x00036C96000 EntryPoint=0x00036C9E354 PciHostBridgeDxe.efi
InstallProtocolInterface: BC62157E-3E33-4FEC-9920-2D3B36D750DF 3596F618
ProtectUefiImageCommon - 0x3596D6C0
  - 0x0000000036C96000 - 0x0000000000010000
SetUefiImageMemoryAttributes - 0x0000000036C96000 - 0x0000000000001000 (0x0000000000004008)
SetUefiImageMemoryAttributes - 0x0000000036C97000 - 0x000000000000D000 (0x0000000000020008)
SetUefiImageMemoryAttributes - 0x0000000036CA4000 - 0x0000000000002000 (0x0000000000004008)
RootBridge: Revision 304
PCIe link not ready (status=8F)

ASSERT_EFI_ERROR (Status = Device Error)
ASSERT [PciHostBridgeDxe] /home/runner/work/RPi4/RPi4/Build/RPi4/DEBUG_GCC5/AARCH64/MdeModulePkg/Bus/Pci/PciHostBridgeDxe/PciHostBridgeDxe/DEBUG/AutoGen.c(445): !EFI_ERROR (Status)
```

Note that this error does not occur with the official release and everything boots fine (without usb so no keyboard nor other usb storages are working).

I'm attaching the full boot log here
[boot.log](https://github.com/pftf/RPi4/files/6892795/boot.log)

TheMindVirus, 2021-07-29:
This is where it is reaching:
https://github.com/tianocore/edk2-platforms/blob/aa3f6fd542e99dde4206537b095f1a2201275e75/Silicon/Broadcom/Bcm27xx/Library/Bcm2711PciHostBridgeLib/Bcm2711PciHostBridgeLibConstructor.c#L207
A check for "Data" parameter is returning 0x8F instead of the expected 0x30.
Is someone playing Bridge Constructor by any chance?

jlinton, 2021-08-18:
There is a set on the ML that will fix this.

jlinton, 2021-09-01:
The .30 firmware that was just released should have fixed this. Depending on your OS you may/may not yet be able to see what is in the PCIe slot (if anything). I'm going to close this, reopen it if the problem persists.

closed 2021-09-01 by jlinton