pftf/RPi4 #168: Boot fails with CONFIG_EFI_DISABLE_PCI_DMA=y

wizeman, 2021-07-31:
When booting a RPi 4 (8G RAM model) with RPi4 UEFI firmware (either with 3G RAM enabled or disabled) + EFI GRUB 2.06-rc1 + Linux kernel version 5.10.54 with `CONFIG_EFI_DISABLE_PCI_DMA` enabled, GRUB seems to fail loading the kernel with the following errors:

```
EFI stub: Booting Linux kernel...
EFI stub: Using DTB from configuration table
EFI stub: Exiting boot services and installing virtual address map...
EFI stub: ERROR: Exit boot services failed
```

The above messages appear and disappear very quickly, being very easy to miss unless you're paying a lot of attention, not to mention that they're basically impossible to read as-is. I had to record a video and watch it frame by frame to be able to read them properly.

And then GRUB says:

```
Failed to boot both default and fallback entries.
```

And then GRUB goes back to the main menu screen.

After searching online, it was suggested that the EFI stub error message is related to the `CONFIG_EFI_DISABLE_PCI_DMA=y` kernel config option.

Specifically, it was suggested that some buggy UEFI firmwares didn't work when that config option was enabled.

To check if indeed that was the problem, it was suggested to boot with the kernel command line parameter `efi=no_disable_early_pci_dma`.

And indeed, when I boot with the above kernel parameter, GRUB loads the kernel without any issues.

It would be good to fix the above issue because it's not obvious that the boot failure is related to that config option (especially considering that the relevant error message passes by so quickly as to almost being imperceptible).

wizeman, 2021-08-29:
> So, I've sorta got to ask, which linux distro/tree had this turned on?

Mine didn't, but I've been running all my machines (including laptops, desktops and servers) with this config option manually turned on for more than a year without any issues (except for this one, of course).

paulwratt, 2022-02-18:
Was this related to the other RPi Kernel XHCI/USB issues from the time period? Is this still an issue with 1.32?

wizeman, 2022-02-18:
> Is this still an issue with 1.32?

I believe so, recently (after having upgraded to 1.32) I had a bootable USB drive in which I had forgotten to add the `efi=no_disable_early_pci_dma` boot parameter and it failed to boot as described in this issue.
It worked once I added the parameter.