pftf/RPi4 #171: Cannot use the GPIOs

luke-par, 2021-08-21:
Hello,

I successfully installed the latest Debian 11 with the Raspberry Pi 4 UEFI firmware v1.29 today.

Now I need access to the GPIOs.
Unfortunately, the access works neither via Python nor directly via sysfs. I also installed the package rpi.gpio-common, but this did not work either.

Do I have to set any parameters in config.txt or does the access not work in general?

TheMindVirus, 2021-08-21:
https://github.com/raspberrypi/windows-drivers/tree/master/drivers/gpio/bcm2836 - This is the driver for GPIO, and
https://github.com/raspberrypi/windows-drivers/tree/master/drivers/i2c/bcm2836 - This is the driver for I2C for HATS.
Most of this predates the Pi4 by a couple of generations and is only made available to C# Namespaces if at all.
The GPIO becomes available with GpioClx (Windows.Devices.Gpio) and
the I2C becomes available with SpbClx (Windows.Devices.Sensors, surprisingly not Windows.Devices.I2c).

I have a half-baked attempt at scrapping both interfaces and using IOCTL (DeviceIoControl) instead:
[I2CTLdev-src.zip](https://github.com/pftf/RPi4/files/7025851/I2CTLdev-src.zip)
It is only a partial merge of the I2C driver with my own IOCTL driver (which I find works well for simple message passing)
(source: https://github.com/TheMindVirus/ioctl) - I really would like to see this used to power HATs and all sorts of things
but it is a pain to develop.

luke-par, 2021-08-21:
Thank you for your reply. Unfortunately it does not help me completely, since I use Debian.

But can you tell me if you have adjusted anything in the UEFI so that you can use the GPIOs under Windows?

TheMindVirus, 2021-08-21:
The only adjustments in UEFI involve changing PinMux routing to a known state at boot, nothing that can't be then changed at runtime within the OS. Debian being Linux-based also has IOCTL but it looks slightly different to Windows IOCTL.
The same thing could apply but doesn't as of yet.

paulwratt, 2022-02-17:
This is not the 1st thread to contain references to the need for GPIO pin management.

Is it possible to load a GPIO text file at run time that presets GPIO.

The idea of auto-detecting HATs is mentioned elsewhere, but what about non-HAT solutions (eg i2c Fans)

Also, some way to lock out (?) OS from specific pin use based on initial settings might be needed (eg pin 14 when used as Fan Control). Is IOCTL_SOCKET a solution here?

space55, 2024-05-03:
Hi,

I wanted to check in and see if in the time between previous messages and now, as to whether or not progress has been made on GPIO from a Pi 4. I have a HAT which requires UART and a pulse-per-second (PPS) pin to be made available. As far as I can tell, GPIO access is not possible in the current state. I am also unable to boot with DeviceTree mode set, it just hangs on launch (though it could be an issue with the distro I'm using).

Thanks