pftf/RPi4 #172: Regression: BootOrder not retained with 1.29 FW

samerhaj, 2021-08-21:
There is a known issue in 1.29 FW that causes BootOrder to not be retained. This is due to: https://github.com/tianocore/edk2-platforms/commit/2e87ce84dc67cd8b2f1a39e34ccdd885a2993226

There is a fix submitted to the mailing list, pending merge: https://edk2.groups.io/g/devel/topic/84967553#79473

Once merged, this will be picked up in the next FW release.

sunnywang-arm, 2021-09-07:
Somehow Greg and I can't reproduce this issue with v1.29. Could someone who ran into this issue help use the bootleg firmware below to get more information for us? I added more debug prints into the bootleg.    
[RPI_EFI_BootOptionTrace.zip](https://github.com/pftf/RPi4/files/7120948/RPI_EFI_BootOptionTrace.zip)

Cc @samerhaj @pbatard

samerhaj, 2021-09-12:
I was able to reproduce the issue using 1.29 and 1.30, but not 1.30 
Also, I confirmed that the 1.31 (which picked up this patch, as well as reverted the older start4/fixup4 binaries) works.

paulwratt, 2022-02-17:
With 1.32 reverting to upstream Kernel again, can this be closed? (did the patchs get accepted - there is no mention of it on the Release page)