pftf/RPi4 #173: GPIO Fan

Kcajminer2312, 2021-08-24:
my GPIO Fan stays running all the time

Kcajminer2312, 2021-08-24:
I need this sorting today,

TheMindVirus, 2021-08-24:
https://www.raspberrypi.org/documentation/computers/raspberry-pi.html#POWER_OFF_ON_HALT

Kcajminer2312, 2021-08-24:
Windows 10 arm, this wont work

TheMindVirus, 2021-08-24:
This is part of firmware, it is underneath Windows 10 arm on the Pi. It is only configurable as of yet with tools from Raspberry Pi OS.

Kcajminer2312, 2021-09-23:
Why did you change the title. That is not what I want

Kcajminer2312, 2021-09-23:
I looked at the convo on discord, you clearly weren't acting mature and people aren't going to put up with that because they have no reason to. They aren't paid for it, it's not a job for them. So why would they waste time and energy when you're only going to act like you're entitled to everything being fixed, immediate support, and everything else this world may be able to offer.

I don’t need this
I was mature
I just need my problems fixing

paulwratt, 2022-02-17:
@Kcajminer2312 _WHAT_ manufacturer? _WHAT_ GPIO pins? _LINK_ to a picture of it connected to GPIO?

Bug Trackers need _DETAILS_ to track bugs. Please supply _DETAILS_. Solution can be provided if _DETAILS_ are provided.

Kcajminer2312, 2022-02-17:
> @Kcajminer2312 _WHAT_ manufacturer? _WHAT_ GPIO pins? _LINK_ to a picture of it connected to GPIO?
> 
> Bug Trackers need _DETAILS_ to track bugs. Please supply _DETAILS_. Solution can be provided if _DETAILS_ are provided.

the fan dosn’t have a manufacturer, I was hoping to get it running on GPIO 3
But this is about a year too late

paulwratt, 2022-02-20:
Nevermind.

At the time it would have helped to at least provide photo or drawing of it connected to RPi (due to lack of manufacturer photo or specs etc)

Kcajminer2312, 2022-02-20:
What the manufacturer got to do with it anyway, it’s the UEFI that dosnt allow you to use any pin

paulwratt, 2022-02-20:
That was half the point.

But, a manufacturer / model _could_ have allowed some research into
their specs, usage etc, _and_ probably would have provided a photo in
some context, especially if the target market was RPi.

A photo could result in the same sort of "issue" as the Official Fan,
where the _packaging photo_ shows Pin 14, but only "for conviniece" of
_also_ showing the other 2 pins. Its things like that you wont see
_reading a data sheet_.

"unknown" is a valid Manufacturer, if you are submitting _details_,
but then you also probably purchased from somewhere, so even if you
dont know the Manufacturer, you can probably still provide a product
link.

There is another Issue listed regarding a RPi, but the photo provided
tells me maybe (upto) 50 different details, _even without a
manufacturer_ reference, _and_ even though a _manufacturer data sheet_
might only show 10 details.

There is a reason for the _old_ saying, "A picture paints a thousand words"

Kcajminer2312, 2022-10-16:
Can this issue be deleted as its the second thing on google images for my name