pftf/RPi4 #174: Version 1.29 No longer boots raspberrypi kernel, only boots upstream kernel

Barborica-Alexandru, 2021-08-24:
Hello, I upgraded today to 1.29 and when trying to boot a manually compiled kernel from https://github.com/raspberrypi/linux I am greeted by this message:
[image]

However the standard upstream kernel provided by fedora boots just fine. Also previous versions of the firmware including 1.28 boots the compiled kernels just fine.

samerhaj, 2021-08-24:
Are you booting in DeviceTree or ACPI mode?

Barborica-Alexandru, 2021-08-24:
This is on ACPI with 3GB limit enforced:
[image]

And this is on pure DeviceTree with 3GB limit enforced:
[image]

Seems the same. Could be possible DeviceTree is not working?

TheMindVirus, 2021-08-25:
CONFIG_SND_HDA_ALIGNED_MMIO=y being missing from the manually compiled kernel might be problematic because the pi4 cannot do unaligned reads and writes without handwritten assembly workarounds. The compiled kernel is probably using unimplemented/erroneous LDUR/STUR instructions and fails to boot.

Barborica-Alexandru, 2021-08-25:
> 
> 
> CONFIG_SND_HDA_ALIGNED_MMIO=y being missing from the manually compiled kernel might be problematic because the pi4 cannot do unaligned reads and writes without handwritten assembly workarounds. The compiled kernel is probably using unimplemented/erroneous LDUR/STUR instructions and fails to boot.

But that flag seems to be related to the intel HDA driver which is not present on the raspberrypi. Anyway I am going to experiment with the flags between the kernels and see what makes it boot. However I think this is a regression since there were no problems up to version 1.28

TheMindVirus, 2021-08-25:
Despite there being no Intel HDA Sound on Raspberry Pi Hardware, the Linux Kernel for it still contains Intel HDA Drivers/Modules in the tree. This could still be one of many potential causes of the boot failure, the others are listed in the changelog for 1.29: Added Secure Boot, Enabled Boot Discovery Policy and Removed "unnecessary" ACPI CID's.

Barborica-Alexandru, 2021-08-25:
Well to eliminate the obvious I copied over the entire config from the fedora kernel over to the raspberrypi kernel and it still does not boot. So there is a change in the source code preventing the raspberrypi kernel from booting on version 1.29.

0n0w1c, 2021-08-25:
I too experience this issue with 5.13 and 5.14. Booting with u-boot and 5.14, also has this issue, so I would guess this is more related to come change in the kernel code than uefi code change.

paulwratt, 2022-02-17:
This is closed? Or is the associated #157 still an open issue as well?

Barborica-Alexandru, 2023-03-13:
I'm going to close this since it's been so long since I ran into this issue that I remember nothing about it.

closed 2023-03-13 by Barborica-Alexandru