pftf/RPi4 #176: The firmware setup doesn't work properly

professzor97, 2021-09-01:
Hello!
I have been trying to install Fling on my Raspberry Pi4b 4gb following instructions in the official pdf downloaded from
https://flings.vmware.com/esxi-arm-edition
with no luck. I can't disable the 3gb ram limit.
After entering the setup by pressing escape I can't go further. The language selection screen appears and the English text sometimes appears and disappears. It does nothing for any input but after a while it starts randomly moving between the menu items like I pressed the arrow keys, then finally it exits and starts to boot. 
I have tryed with a Genius wireless keyboard and with a usb keyboard, both of them were tried in 2.0 and 3.0 usb ports of the pi, and in an active hub with the same result. I have also tryed another version of the Uefi firmware. 
I am very interested in the Fling project and I would like to see what it is capable of and it is very frustrating that I can't pass this problem.
Thanks for any help!

jlinton, 2021-09-01:
Do you have something plugged into the RPI/GPIO/expansion port? I saw something similar to this last week when I lost the ground line on my serial->USB adapter and it was blasting random garbage into the rpi's serial port. Granted linux/etc but the behavior was similar, the UEFI menus/etc would jump around as I used the graphical head+USB keyboard.

professzor97, 2021-09-03:
I have a Pimoroni Fan Shim. I disconnected it from the gpio yesterday and surprisingly it worked! 
Thanks for the help, I would never thought that a gpio device, mainly a fan could cause the problem!

closed 2021-09-03 by professzor97

jlinton, 2021-09-03:
Ah, yah its the LED on that board probably causing the problem. If you assure that GPIO 14/15 are disconnected I think you can probably still use the fan. There is an opinion in the UEFI menus to use GPIO 18 as used on that board for the fan controller. The LED's are tied to the serial TX/RX data lines so basically it's probably dumping some amount of TX data into the RX and causing the menus to jump around.