pftf/RPi4 #177: v1.30 not booting on Raspbbery Pi 4 8GB

stgraber, 2021-09-04:
I've been trying to get this UEFI firmware working on my RaspberryPi for a little while but as I need netbooting, I've had very limited luck so far :)

With this release supposedly fixing that, I was quite excited to try it again, sadly it's working even less than the previous release.
Booting v1.29, I get EDK2 starting, can access the firmware menu, ... I just can't boot from the network.

With v1.30, the screen flashes a couple of time (remaining blank though) and then just sits with nothing on it, no boot logo, no EDK2 firmware menu, nothing.

I hooked up a TTL adapter and I'm getting this set of boot messages (have rpi-eeprom configured to do UART):
```
PM_RSTS: 0x00001000
RPi: BOOTLOADER release VERSION:c2f8c388 DATE: Apr 29 2021 TIME: 17:11:29 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=
1619712685 0x9f3b0cac 0x00d03114 0x000c4304
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 3200
PCI reset
PCI reset
VLI: HUB2: 0xfff00000 0x24e6 MCU: 0xfff20000 0x15218
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
Boot mode: SD (01) order f4
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
OCR c0ff8000 [17]
CID: 00035344534c333247801ced040a0141
CSD: 400e00325b590000edc87f800a404000
SD: bus-width: 4 spec: 2 SCR: 0x02358043 0x00000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
MBR: 0x00000001,62333951 type: 0xee
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
GPT 4b6635133a7fda408b3cf0cb919b0818 000000001 003b723ff num-partitions 128 entry-size 128
GPT partition: 0 28732ac11ff8d211ba4b00a0c93ec93b 000000800..0004007ff
lba: 2048 oem: 'mkfs.fat' volume: ' NO NAME    '
rsc 32 fat-sectors 4088 c-count 523262 c-size 8 r-dir 2 r-sec 0
PM_RSTS: 0x00001000
Trying partition: 0
GPT 4b6635133a7fda408b3cf0cb919b0818 000000001 003b723ff num-partitions 128 entry-size 128
GPT partition: 0 28732ac11ff8d211ba4b00a0c93ec93b 000000800..0004007ff
lba: 2048 oem: 'mkfs.fat' volume: ' NO NAME    '
rsc 32 fat-sectors 4088 c-count 523262 c-size 8 r-dir 2 r-sec 0
Read config.txt bytes      206 hnd 0x000002c3
Read start4.elf bytes  2240352 hnd 0x000002c6
Read fixup4.dat bytes     5407 hnd 0x000002c4
Firmware: 67615e950e1e28b92dfae6303cf7a8b879a8908f Aug 31 2021 14:52:47
0x00d03114 0x00000000 0x00000fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0
PCI reset
+

MESS:00:00:03.692942:0: arasan: arasan_emmc_open
MESS:00:00:04.187970:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:04.190752:0: brfs: File read: 206 bytes
MESS:00:00:04.285736:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:04.295258:0: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:04.304765:0: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:04.314279:0: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:04.323789:0: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:04.333303:0: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:04.342810:0: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:04.352324:0: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:04.361834:0: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:04.371349:0: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:04.375848:0: HDMI1:EDID giving up on reading EDID block 0
[... 144 lines cut ...]
MESS:00:00:05.749895:0: HDMI1: hdmi_pixel_encoding: 300000000
MESS:00:00:05.761036:0: dtb_file 'bcm2711-rpi-4-b.dtb'
MESS:00:00:05.775069:0: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
MESS:00:00:05.778355:0: Loading 'bcm2711-rpi-4-b.dtb' to 0x1f0000 size 0xc2a1
MESS:00:00:05.816973:0: brfs: File read: 49825 bytes
MESS:00:00:05.980694:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.983417:0: brfs: File read: 206 bytes
MESS:00:00:05.988928:0: brfs: File read: /mfs/sd/overlays/miniuart-bt.dtbo
MESS:00:00:06.044589:0: Loaded overlay 'miniuart-bt'
MESS:00:00:06.179865:0: brfs: File read: 1819 bytes
MESS:00:00:06.181851:0: Failed to open command line file 'cmdline.txt'
MESS:00:00:06.468787:0: brfs: File read: /mfs/sd/RPI_EFI.fd
MESS:00:00:06.471253:0: Loading 'RPI_EFI.fd' to 0x0 size 0x1f0000
MESS:00:00:06.477079:0: No compatible kernel found
MESS:00:00:06.481581:0: Device tree loaded to 0x1f0000 (size 0xc818)
MESS:00:00:06.489229:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:06.496730:0: uart: Baud rate change done...
MESS:00:00:06.498753:0: uart: Baud rate change done...
MESS:00:00:06.503942:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
NOTICE:  BL31: v2.5(release):
NOTICE:  BL31: Built : 18:50:30, Jun 14 2021
UEFI firmware (version UEFI Firmware v1.29 built at 15:45:19 on Aug  4 2021)

ESC (setup), F1 (shell), ENTER (boot)......
```

stgraber, 2021-09-04:
And did a run with the debug build of v1.30:
```
PM_RSTS: 0x00001000
RPi: BOOTLOADER release VERSION:c2f8c388 DATE: Apr 29 2021 TIME: 17:11:29 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1619712685 0x9f3b0cac 0x00d03114 0x000c424f
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 3200
PCI reset
PCI reset
VLI: HUB2: 0xfff00000 0x24e6 MCU: 0xfff20000 0x15218
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
Boot mode: SD (01) order f4
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
OCR c0ff8000 [17]
CID: 00035344534c333247801ced040a0141
CSD: 400e00325b590000edc87f800a404000
SD: bus-width: 4 spec: 2 SCR: 0x02358043 0x00000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
MBR: 0x00000001,62333951 type: 0xee
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
GPT 4b6635133a7fda408b3cf0cb919b0818 000000001 003b723ff num-partitions 128 entry-size 128
GPT partition: 0 28732ac11ff8d211ba4b00a0c93ec93b 000000800..0004007ff
lba: 2048 oem: 'mkfs.fat' volume: ' NO NAME    '
rsc 32 fat-sectors 4088 c-count 523262 c-size 8 r-dir 2 r-sec 0
PM_RSTS: 0x00001000
Trying partition: 0
GPT 4b6635133a7fda408b3cf0cb919b0818 000000001 003b723ff num-partitions 128 entry-size 128
GPT partition: 0 28732ac11ff8d211ba4b00a0c93ec93b 000000800..0004007ff
lba: 2048 oem: 'mkfs.fat' volume: ' NO NAME    '
rsc 32 fat-sectors 4088 c-count 523262 c-size 8 r-dir 2 r-sec 0
Read config.txt bytes      206 hnd 0x000004b4 
Read start4.elf bytes  2240352 hnd 0x000004b7 
Read fixup4.dat bytes     5407 hnd 0x000004b5 
Firmware: 67615e950e1e28b92dfae6303cf7a8b879a8908f Aug 31 2021 14:52:47
0x00d03114 0x00000000 0x00000fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0
PCI reset
+

MESS:00:00:03.692215:0: arasan: arasan_emmc_open
MESS:00:00:04.187010:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:04.189792:0: brfs: File read: 206 bytes
MESS:00:00:04.284762:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:04.294278:0: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:04.303786:0: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:04.313300:0: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:04.322810:0: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:04.332324:0: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:04.341831:0: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:04.351345:0: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:04.360855:0: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:04.370369:0: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:04.374868:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:04.382388:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.472515:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:05.479809:0: *** Restart logging
MESS:00:00:05.481233:0: brfs: File read: 206 bytes
MESS:00:00:05.522910:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.532951:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:05.549632:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:05.559727:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
[... 1408 lines cut ...]
InstallProtocolInterface: BC62157E-3E33-4FEC-9920-2D3B36D750DF 35982918
ProtectUefiImageCommon - 0x359825C0
  - 0x0000000036AA9000 - 0x000000000000A000
SetUefiImageMemoryAttributes - 0x0000000036AA9000 - 0x0000000000001000 (0x0000000000004008)
SetUefiImageMemoryAttributes - 0x0000000036AAA000 - 0x0000000000007000 (0x0000000000020008)
SetUefiImageMemoryAttributes - 0x0000000036AB1000 - 0x0000000000002000 (0x0000000000004008)
InstallProtocolInterface: 18A031AB-B443-4D1A-A5C0-0C09261E9F71 36AB1078
InstallProtocolInterface: 107A772C-D5E1-11D4-9A46-0090273FC14D 36AB10D8
InstallProtocolInterface: 6A7A5CFF-E8D9-4F70-BADA-75AB3025CE14 36AB10F0
Loading driver 100C2CFA-B586-4198-9B4C-1683D195B1DA
InstallProtocolInterface: 5B1B31A1-9562-11D2-8E3F-00A0C969723B 359762C0
add-symbol-file /home/runner/work/RPi4/RPi4/Build/RPi4/DEBUG_GCC5/AARCH64/Platform/RaspberryPi/Drivers/ArasanMmcHostDxe/ArasanMmcHostDxe/DEBUG/ArasanMMCHost.dll 0x36AA2000
Loading driver at 0x00036AA1000 EntryPoint=0x00036AA4AA0 ArasanMMCHost.efi
InstallProtocolInterface: BC62157E-3E33-4FEC-9920-2D3B36D750DF 35976A18
ProtectUefiImageCommon - 0x359762C0
  - 0x0000000036AA1000 - 0x0000000000008000
SetUefiImageMemoryAttributes - 0x0000000036AA1000 - 0x0000000000001000 (0x0000000000004008)
SetUefiImageMemoryAttributes - 0x0000000036AA2000 - 0x0000000000005000 (0x0000000000020008)
SetUefiImageMemoryAttributes - 0x0000000036AA7000 - 0x0000000000002000 (0x0000000000004008)
SD is routed to emmc2
InstallProtocolInterface: 3E591C00-9E4A-11DF-9244-0002A5F5F51B 36AA7030
InstallProtocolInterface: 964E5B21-6459-11D2-8E39-00A0C969723B 359767C8
InstallProtocolInterface: 09576E91-6D3F-11D2-8E39-00A0C969723B 35976698
ASSERT [ArasanMMCHost] /home/runner/work/RPi4/RPi4/edk2-platforms/Platform/RaspberryPi/Drivers/ArasanMmcHostDxe/ArasanMmcHostDxe.c(263): BaseFrequency != 0
```

stgraber, 2021-09-04:
```
ASSERT [ArasanMMCHost] /home/runner/work/RPi4/RPi4/edk2-platforms/Platform/RaspberryPi/Drivers/ArasanMmcHostDxe/ArasanMmcHostDxe.c(263): BaseFrequency != 0
```

Obviously looks like a bit of a problem :)

stgraber, 2021-09-04:
Hopefully that helps tracking this down. Let me know if you have a tentative fix you'd like me to test as that should be pretty straightforward for me to do now.

stgraber, 2021-09-04:
Just for kicks since MMC was mentioned in there, I tried booting from USB with the same result. That makes sense as it looks like it's the MMC driver blowing up and it'd be loading regardless.

stgraber, 2021-09-04:
This may be a DTB issue or something similar as taking the v1.29 release and only replacing the EDK2 firmware itself, keeping everything else from v1.29 seems to work just fine here (including netboot).

sunnywang-arm, 2021-09-05:
@stgraber 

Yeah, I ran into this issue as well.  

The reason why the v1.30 release build can boot is that the ASSERT doesn't exist (ASSERT is not built into the release build). However, the v1.30 release build would still run into the SD/eMMC2 (SD card) related issue. For example, the variable store doesn't work, so BootOrder and all setup options can't be changed. 

I'm using a similar solution to yours now, but my base is v1.28.  Using "RPi4_UEFI_Firmware_v1.28 files and then only replace RPI_EFI.fd with the firmware that you want to use (the one in RPi4_UEFI_Firmware_v1.30 or the one that is built from edk2 TOT). Then, it works. 

The different files between v1.28 and v1.30 are the following. I'm not sure which file/files cause this issue, and will spare time to narrow this down. 
- bcm2711-rpi-4-b.dtb
- bcm2711-rpi-400.dtb
- bcm2711-rpi-cm4.dtb
- fixup4.dat
- start4.elf

Add @pbatard @ardbiesheuvel @samerhaj  @jlinton  They may have a quick idea.

jlinton, 2021-09-06:
This is likely hitting everyone, the upstream pi firmware is having issues everywhere: https://github.com/raspberrypi/firmware/issues/1619

jlinton, 2021-09-06:
I think for the time being we should probably roll back to an earlier start4/fixup4.

jlinton, 2021-09-07:
I can't reproduce this today to save my life, I guess it must be affected by something else in the machine? But there _is_ a bug, which could cause the base frequency problem. Although how changing the start4/etc firmware might trigger it, isn't clear.

RpiFirmwareGetClockRate() needs to have   

ReleaseSpinLock (&mMailboxLock); 

moved into the failure case and below the 

*ClockRate = Cmd->TagBody.ClockRate; 

line.

Without that, it might be possible for some other driver/etc to issue a mailbox command and clear the return data (AFAIK) since it is a shared buffer.

jlinton, 2021-09-10:
It also looks like we might be able to just hardcode the request frequency, request it, and ignore the return from getclock rate. Given we really need those values to be 250 (EMMC) and 100mhz (EMMC2) for Linux to work, it might not be a bad temporary fix.  Doing that and the machine seems to work properly.

Also, @stgraber can you tell me the man/model/etc of the card that is failing?

stgraber, 2021-09-10:
The card is a SanDisk Edge 32GB SDHC A1, it's what Canakit was bundling with their RaspberryPi4 kits at the time I ordered a few of those.

paulwratt, 2022-02-17:
Is this fixed now (with the update  - and the Kernel updates since this was last posted)? or is the workaround still needed (based on SD-Card failure)?

closed 2022-03-07 by pbatard