pftf/RPi4 #178: Raspberry Pi 4 UEFI won't boot

ChibiANUBIS, 2021-09-04:
Hello my dear, I reinstall Windows 10 on UEFI in my SD Card, I use the WoR tool,
But my monitor display "No signal" and my Raspberry yellow LED display 4 times,

Can you help please,

Regards.

sunnywang-arm, 2021-09-06:
@ChibiANUBIS 
I would suggest that you just simply unzip the RPi4_UEFI_Firmware_v1.27.zip in https://github.com/pftf/RPi4/releases/tag/v1.27 to the SD card's partition. There are some issues with v1.28, and later, so v1.27 may be a better choice for you at this moment. 
By the way, the partition should be formatted in FAT32. If it works, it may be something wrong with the partitions and files generated by the WoR tool rather than the UEFI firmware itself. For WoR related issues, it would be better to go to the community or bug system: 
https://www.worproject.ml/community
https://www.worproject.ml/bugtracker/my_view_page.php

ChibiANUBIS, 2021-09-06:
Thanks my dear for your help, but is the same, Is not possible for me to run anything

sunnywang-arm, 2021-09-07:
@ChibiANUBIS 
Could you connect the serial console to get the serial log?

ChibiANUBIS, 2021-09-08:
> @ChibiANUBIS
> Could you connect the serial console to get the serial log?

No, what is this ?

sunnywang-arm, 2021-09-08:
@ChibiANUBIS 
Here you go.
https://learn.adafruit.com/adafruits-raspberry-pi-lesson-5-using-a-console-cable/overview

ChibiANUBIS, 2021-09-08:
> @ChibiANUBIS
> Here you go.
> https://learn.adafruit.com/adafruits-raspberry-pi-lesson-5-using-a-console-cable/overview

Is not possible to enable with the WoR...

sunnywang-arm, 2021-09-10:
> > @ChibiANUBIS
> > Here you go.
> > https://learn.adafruit.com/adafruits-raspberry-pi-lesson-5-using-a-console-cable/overview
> 
> Is not possible to enable with the WoR...

@ChibiANUBIS 
Hmm... I don't understand. 
Even if you use WoR to produce the UEFI boot partition and UEFI firmware, you should still be able to enable the serial console by using the steps mentioned in the link and a USB To TTL cable. We just want to get serial console output from firmware rather than OS.
- https://learn.adafruit.com/adafruits-raspberry-pi-lesson-5-using-a-console-cable/you-will-need
- https://learn.adafruit.com/adafruits-raspberry-pi-lesson-5-using-a-console-cable/connect-the-lead
- https://learn.adafruit.com/adafruits-raspberry-pi-lesson-5-using-a-console-cable/test-and-configure

closed 2021-11-22 by ChibiANUBIS