pftf/RPi4 #179: Boot order is not getting set automatically correctly since 1.28

Marcinoo97, 2021-09-05:
Some people in WoR server reported that their pi 4 defaults to http boot while in the past it was always defaulting to the device uefi got booted from.

pbatard, 2021-09-07:
Yeah, I've been seeing that issue too when installing Debian 11 on SD card (boot order kept being reverted to PXE having highest priority, which is annoying).

Obviously, the most likely culprit is tianocore/edk2-platforms@2e87ce84dc67cd8b2f1a39e34ccdd885a2993226 that was introduced in 1.29...

pbatard, 2021-09-07:
At any rate, this is a duplicate of #172, so I'm just going to keep that last issue open.

closed 2021-09-07 by pbatard

sunnywang-arm, 2021-09-07:
@pbatard 
Yeah, I think so. This issue should be caused by tianocore/edk2-platforms@2e87ce84dc67cd8b2f1a39e34ccdd885a2993226. 
However, this looks like a different issue from #172. This issue looks like talking about the default boot order rather than BootOrder change preservation. Do I miss anything here? If not, we may still need to keep this open for discussing the default boot order issue/support. 
Cc @Marcinoo97

sunnywang-arm, 2021-09-07:
@pbatard 

> Yeah, I've been seeing that issue too when installing Debian 11 on SD card (boot order kept being reverted to PXE having highest priority, which is annoying).
> 

Which firmware version did you use? If you used v1.30, it may be caused by the problem discussed in #177 rather than tianocore/edk2-platforms@2e87ce84dc67cd8b2f1a39e34ccdd885a2993226. Actually, I can't reproduce the Boot Order change preservation issue(#172) on my system with v1.29. @samerhaj, Greg, and I are sparing time to take a further look.

pbatard, 2021-09-09:
@sunnywang-arm, I've been using 1.30, and I have now produced release v1.31, that includes the non-integrated patch submitted at https://edk2.groups.io/g/devel/topic/84967553#79472 and (from some very limited testing) it seems to fix my issue with the boot order not being preserved.

sunnywang-arm, 2021-09-10:
> @sunnywang-arm, I've been using 1.30, and I have now produced release v1.31, that includes the non-integrated patch submitted at https://edk2.groups.io/g/devel/topic/84967553#79472 and (from some very limited testing) it seems to fix my issue with the boot order not being preserved.

@pbatard 
Do you mean you see the boot order issue with v1.30? if so, it probably is caused by start4.elf/fixup.dat change rather than the edk2 code change. I can reproduce the boot order issue with v1.30 too, but can't reproduce it with v1.29. 
If you see the boot order issue with v1.29, could you use my bootleg/test firmware in https://github.com/pftf/RPi4/issues/172#issuecomment-914206647 to reproduce the issue and get the log for me?

samerhaj, 2021-09-12:
FWITW, I have confirmed that 1.31 works (on RPi400 at least). I am able to retain the USB HDD at the top of the BootOrder list. Going back to 1.30 or 1.29 on the same Setup does not work, while 1.28 works.