pftf/RPi4 #180: Argon One case and fan setting

sHAsHiLx, 2021-09-07:
Good day,

What settings will fit for this case? Pin 18 or 19?

jlinton, 2021-11-13:
This is variable speed via i2c? One of the things here is to start porting some of these drivers to the efi driver model. Because, yes, ACPI defines the thermal trip points that the OS is going to respond to, as well as adjusting fan speeds. AKA its a standardized fan control mechanism that doesn't require every single fan to have its own daemon. The rpi makes putting fans on odd busses easy but completely one off, unlike the 4 pin fan model you get on most x86 boards.

Killertamagotchi, 2021-12-15:
This is the pinout from the Argon One Case

GPIO PIN | Usage

7 | Shut Down Monitor (Monitor Cut if No Signal)
11 | Power Button
22 | IR Transmit
23 | IR Receive
27/28 (i2c) | FAN Control via i2C

So yes the FAN uses i2c

https://github.com/Argon40Tech/Argon-ONE-i2c-Codes

paulwratt, 2022-02-17:
@Killertamagotchi in creating a standard i2c Fan Control Driver, some way to define various pins is required, that would be the only way to release something that works across all platforms and all i2c Fan Control solutions. That said something for Power Button, monitor Toggle and IR Control would also have a need for such defines too (I am also thinking of Suspend & Sleep). It maybe that a GPIO control file might be used, to populate settings. RPi4 has a default Reset on Power Off GPIO pin combo (that probably works on CM4 too - maybe carrier board dependant tho).

Anyways, there is apparently _no_ driver atm.

sallevan, 2024-03-25:
Just a correction to clarify GPIO quantity occupied by Argon ONE.

> This is the pinout from the Argon One Case
> 
> GPIO PIN | Usage
> 
> 7 | Shut Down Monitor (Monitor Cut if No Signal)
> 11 | Power Button

Not exactly that.

Raspberry Pi header pin 7, which is GPIO.7 in terms of Argon and GPIO4 in terms of Raspberry Pi is ... o.k, "an industry de-facto standard" for Power button. Many UPS hats use it that way.

Raspberry Pi header pin 8, which is TxD0 in terms of Argon and Raspberry Pi (alternatively -- GPIO14 in Raspberry) is a pin which is monitored for platform reboot.
Being UART transmit pin, it becomes high after UART initialisation because COM port uses inverse logic, so, "no signal" is "logic one", but goes low when BCM resets. Argon (and many other Raspberry Pi power control appliances) translate this to "power off" event if they have been programmed previously to do so.

**No additional GPIO pin is required,** look at the picture on [Waveshare Wiki](https://www.waveshare.com/wiki/PI4-CASE-ARGON-ONE).

> 22 | IR Transmit
> 23 | IR Receive
> 27/28 (i2c) | FAN Control via i2C

Once again, Argon One doesn't ... o.k, "steal" these I2C pins, it is "just another I2C device on the bus", so, you can use them to attach your favourite I2C devices unless they clash with Argon One ID (0x1a).

sallevan, 2024-03-25:
BTW, there is an unobvious question why Argon NEO Raspberry Pi case uses Raspberry Pi header pin 5 (SCL.1 or GPIO3) for its Power button -- PCB tracing error?
Pressing it effectively blocks main I2C bus for transactions.

paulastronaut, 2025-05-31:
Has anybody got the Argon One case working with UEFI boot? Like @0n0w1c I want to use the daemon from [https://gitlab.com/DarkElvenAngel/argononed](url), but the daemon can't connect to I2C because it is not available. Currently, I am trying to figure out how to enable I2C on the RPi while using UEFI boot. It seems to me that this is not possible at the moment (e. g. see [https://github.com/pftf/RPi4/issues/130](url)), but perhaps my conclusion is wrong. Any hints would be welcome.