pftf/RPi4 #181: Serial over USB

damfle, 2021-09-11:
AFAIK Pi4 can act as some USB device. 
It would be nice if the firmware could have the USB C port acting as USB-UART serial port. 
This would allow simplify management of clusters of Pi (without PXE boot).

TheMindVirus, 2021-09-12:
I had done this for a Pi4-based OMX Video Player system once, it revolves around the dwc2 module for Firmware and Linux.
Not sure about Windows support for dwc2 but it at least has COM1 for the GPIO UART RX/TX pins (which may have rerouting).

/boot/config.txt
```
...
dtoverlay=dwc2 #this may have changed a bit
enable_uart=1
uart_2ndstage=1
```

/boot/cmdline.txt
```
console=serial0,115200 consoleblank=0 root=PARTUUID=283dc939-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait modules-load=dwc2,g_ether quiet splash loglevel=0 logo.nologo vt.global_cursor_default=0 plymouth.enable=0 plymouth.ignore-serial-consoles fastboot
```
* root=PARTUUID may have to change to match the actual OS partition. A lot of this isn't available for editing in UEFI and would also be nice if it included a basic text editor for modifying any boot partition file. This would save a lot of adding and removing options later.

concussious, 2025-05-27:
I can't get USBC serial to work for EDK2 itself on a Pi400, with no OS, just an EFI directory on MicroSD. Is it possible to do this?

The EFI seems to be otherwise working on Pi400 and does boot a USB.