pftf/RPi4 #187: PoE-Plus

tahikib, 2021-10-19:
I have the official Raspberry Pi PoE+ hat [https://www.raspberrypi.com/products/poe-plus-hat/](https://www.raspberrypi.com/products/poe-plus-hat/)

I have firmware 1.32 and have tried 1.29 Hybird RPI_EFI.fd and firmware 1.31.
I cannot get the fan to kick on at all on this hat.

I have tried modifying the config.txt with
```
# PoE Hat Fan Speeds
dtoverlay=rpi-poe-plus
dtparam=poe_fan_temp0=10000,poe_fan_temp0_hyst=1000
dtparam=poe_fan_temp1=55000,poe_fan_temp1_hyst=5000
dtparam=poe_fan_temp2=60000,poe_fan_temp2_hyst=5000
dtparam=poe_fan_temp3=65000,poe_fan_temp3_hyst=5000
```

I had tried changing the GPIO from Fanshim GPIO 18 and GPIO 19 setting the temp to 50 and it will not come on.

I am trying to use ESXi by Fling for ARM.

Am I doing something wrong or is the UEFI not able to turn on the fan at all regardless of the OS?

buckaroogeek, 2021-10-20:
Doesn't the OS need to also provide a kernel driver? Raspberry OS includes those drivers out of the box but other distributions may or may not have merged those drivers into their kernel code base. As of linux kernel 5.13 there is a separate driver for the original POE Hat fan but it may not work for the poe plus hat. Does Fling ESXi include the Raspberry or upstream drivers?

tahikib, 2021-10-20:
> First thing to check is if your boot partition has a folder called "overlays" containing a file called "rpi-poe-plus" inside it. Second thing to check is how new your "start.elf" is, both that and the Pi4 EEEPROM version may have been from before the PoE+HAT was released. ESXi will also require 4GB RAM so on a 4GB Pi gpu_mem must be set to 0.

Also if you couldn't' tell I am new to this.
I do have both the rpi-poe and rpi-poe-plus overlay.  I have the enable 3GB limit disabled as it is 8gb rpi 4.  The fan turns on when running Raspbian already. 

  I am not sure if Fling does include the drivers.  Is there a way I can check?