pftf/RPi4 #192: Cannot boot Windows from eMMC on CM4

halentech, 2021-11-05:
Still not working on NEC CM4 with version 1.32

jlinton, 2021-11-13:
Hmm, I'm not sure why this would be different from the normal pi4. Have you tried changing the sd controller to the arasan, "Device Manager"->"Raspberry Pi Configuration"->"SD/MMC configuration"->"uSD/eMMC Routing"

 Do you have a serial port what does it print during boot?

I assume this is windows doesn't boot, not that uefi doesn't get windows starting...

halentech, 2021-11-16:
Those screen I got
[image]
[image]
[image]

jlinton, 2021-11-29:
If you hit esc, do and go into the boot device area, do you see windows listed, how about if you drop to the shell, and start the boot manager directly via bootaa64.efi.