pftf/RPi4 #193: Fedora 35 and Device Tree

buckaroogeek, 2021-11-07:
I installed Fedora 35 minimal to a USB drive and then replaced the contents of the EFI partition with v 1.32 release. I changed the EFI partition type to EFI but no other changes. Attached drive to an RPi4, attached hdmi cable and keyboard and booted.  In the UEFI manager menu, disabled the 3 GB limit and changed the system table setting to Device Tree. 

Device booted without issue.  /proc/device-tree is populated which is excellent. I have not seen the device tree in several past releases of the UEFI firmware.

I added upstream_kernel=1 to config.txt and rebooted. The same contents showed up in /proc/device-tree which indicates to me that the upstream device tree was not being parsed. Should the upstream device tree have been read instead (in fedora this is located at /boot/dtb/broadcom)?  

I next uploaded a custom device tree to /boot/dtb/broadcom that turns on the POE Hat fan. Again ignored. Copied the file to /boot/efi and rebooted. The POE hat fan came on.

Should the upstream_kernel=1 setting work and load the upstream device tree is present? Or am I missing some fundamental?  I am really glad to the device tree working with UEFI however. 

Now if i could only get the POE Hat fan working via ACPI ....

thanks!

jlinton, 2021-11-13:
Well the one thing that works is editing the config.txt and adding overlay directives with dtoverlay. Those dto files are in an overlay directory on the sd/usb disk device you are booting. For the most this is the right way to do DT's with this firmware since the start4.efi, tfa and uefi all tweak them in various ways as the machine is booted. 

The kernel provided DT's are based on what is slowly becoming a historical artifact. I would expect that as people try to firm up how they are defined and used (because right now they are basically just reflections of the architecture of the linux kernel).

So, there is a framework and example code on how to use the POE hat in ACPI mode here: https://github.com/jlinton/edk2-platforms/commits/cppc6-various-fixes-vc4aml-mailboxspinlock-xhcipci-clocktweaks
and here https://github.com/jlinton/arm-trusted-firmware/commits/rpi4_cppc_pci_poe2

All that said, the fan works but isn't visible as a ACPI fan object simply by adding the thermal trip points/etc in the config.txt along with the appropriate overlay.

This is all in the works but depends on me eventually getting motivated to push the cppc/pcc/mailbox stuff to TFA, which is likely an uphill battle because its ACPI only.

buckaroogeek, 2021-11-13:
Thanks for the insights Jeremy. They are very helpful.  

My particular use case for the RPi is pretty narrow: running a small kubernetes cluster using Fedora as the OS. The RPi4 has a good balance of performance and cost, with kubernetes enabling hardware redundancy which is useful for situations like mine: small agriculture operation that has limited (and expensive) wireless bandwidth connection to the internet.  The POE Hat is useful in that each RPi4 can be cooled based on the workload the SOC is performing in the cluster at the moment. Their seems to be a match between the goals of the UEFI project (sbbr server compliant) and this use case hence my interest in UEFI. I am also exploring Fedora CoreOS as a replacement for Fedora minimal I currently use.

So for me the upstream device tree is suitable especially since it can support the POE Hat fan. I know I could avoid all of this by just using Raspian OS but I have been a Fedora user for decades. 

Nicolas Saenz Julienne wrote the upstream Hat kernel module as a PWM device which seems to be appropriate. I readily admit however to not being all that familiar with lower level kernel and hardware interfaces. I speculate that using PWM would facilitate the switch from device tree to acpi but that just reflects by lack of detailed understanding of all things kernel related.

I tried to write an overlay for the upstream POE hat kernel module using a tip from Nicolas but cannot get the fan working. Probably a coding error on my part. I have the dtbo file in /boot/efi/overlays. 

In any case, the fan is currently working with Fedora and Fedora CoreOS on the RPI4 along with the UEFI boot loader. It does seem a bit fragile so I look forward to any improvements and alternatives that come along. The workaround to not having the upstream_kernel=1 setting in config.txt  not making a difference is to copy the upstream device tree into /boot/efi.

Best wishes

jlinton, 2021-11-29:
Re-reading this, I'm not sure I was clear, but look at this commit, which hasn't yet rolled into the officially released images here:

https://github.com/pftf/RPi4/pull/188/files

It uses the rpi foundation provided DT's but uses their upstream overlay which should enable some of the upstream DT functionality you are looking for without using the upstream DT.

buckaroogeek, 2021-11-29:
Thank you very much Jeremy. I will indeed take a close look.

Best regards

closed 2021-12-06 by buckaroogeek