pftf/RPi4 #196: possible info on ACPI/device tree

frezbo, 2021-11-25:
What's the difference between using ACPI/devicetree/ACPI+device tree?

Would one provide an advantage over the other in the case of Pi. Reading through some docs it's confusing and ARM seems to prefer device tress. Seems Pi PoE hats require devicetree to work. Does PSCI work if only device tree is used?

jlinton, 2021-11-29:
The dual ACPI+DeviceTree option on the rpi4 provides both table types to the OS, and the OS chooses one or the other. Which one gets chosen is OS-dependent. The firmware options for ACPI or DT only provide one of the two sets of tables, but they also may slightly tweak machine behavior to support that table type.

So with linux when the RPI is in ACPI+DT mode, passing acpi=force, or acpi=off then tells Linux to use ACPI or ignore it. Most but not all disto/linux kernels when faced with this situation will pick DT by default.

So, POE hats work, either way, control over the fan speeds is selected at boot time via config.txt and its DT overlays even when in ACPI mode (except if your running some of my out of tree patches which wrap the POE fan in an ACPI fan object). 

PSCI is used by both ACPI and DT.

For the most part, unless you have a solid reason on this platform for running DT you're probably better off with ACPI. If something happens to not work (like the VC, audio, etc) we welcome contributions as those parts are slowly being enabled for ACPI. If you have a custom piece of HW wired up via GPIO/SPI/etc it's probably better to use DT because there aren't really any plans to export some of the lower level peripheral types in ACPI mode, rather reserving them for firmware uses.

buckaroogeek, 2021-11-29:
Apologies for jumping into this thread. The information Jeremy posted very useful. Is there an online write up or note that shows what is needed to manage the official POE Hat fan via ACPI instead of the DT? I have tried searching but nothing appears to be showing up. I would be glad to do a write up if I can get some pointers.

best regards

closed 2021-12-06 by jlinton