pftf/RPi4 #197: Document .zip defaults versus F9 defaults?

bconway, 2021-12-01:
Greetings. It would be great if the Readme.md or another doc indicated how default settings are meant to work. I noticed that hitting F9 changes a *lot* of stuff, including SDHC mode and 3/4 GB setting. Should I not be hitting F9 when using the RPi 4? Should F9 be updated to use the preferred defaults for the RPi as shipped in the .zip? Any thoughts (or updates) appreciated.

jlinton, 2021-12-02:
Hi, I think what you are seeing is the fact that the HII menus have default values indicated on their HII form fields, but on the first boot, or via the build, we have different defaults depending on the platform (rpi3, rpi4, cm4, installed memory, etc). 

If you hit F9, what you see is the generic HII menu default rather than the platform default you get after installing a new/clean firmware and booting it.

So, in summary, yes. The defaults you get after hitting F9, probably aren't the same as the ones you get with a clean firmware image after its first boot. I'm not really sure how to fix that at the moment. Some of the rpi3 vs 4 things could be tweaked in the menus via #defines to reflect the ConfigDxe compile and runtime defaults, but that creates a bit of a maintenance issue because those would need to be kept in sync by hand to avoid the problem your seeing. The runtime selected defaults are probably a bit trickier because the HII definitions also need to be updated at runtime. Maybe it is best not to use F9 if you expect it to reset to first boot defaults rather than a secondary set of defaults. OTOH, those defaults are selected as the lowest common denominator that boots the largest number of OS's out of the box, they are usually pretty poor selections in general if you're running a recent Linux/etc. The 3G limit option being the prime example.

bconway, 2021-12-02:
> Maybe it is best not to use F9 if you expect it to reset to first boot defaults rather than a secondary set of defaults.

Thank you, that answers my question. Maybe worthy of a note in the Readme.md's Additional Notes section?

closed 2023-11-18 by bconway