pftf/RPi4 #201: Boot stuck for 10 minutes if two external USB 3.0 storage devices are connected

frezbo, 2021-12-08:
serial console

```bash
Read start4.elf bytes  2243232 hnd 0x00002e1d 
Read fixup4.dat bytes     5351 hnd 0x00002e12 
Firmware: 0403e22018aafab833d0a16374ac773f66fd7be9 Oct 19 2021 11:50:10
0x00d03114 0x00000000 0x00000fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0
PCI reset
+

MESS:00:00:07.536887:0: arasan: arasan_emmc_open
MESS:00:00:07.967405:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:07.970195:0: brfs: File read: 206 bytes
MESS:00:00:08.024457:0: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:08.033973:0: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:08.043481:0: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:08.052995:0: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:08.062505:0: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:08.072020:0: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:08.081528:0: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:08.091043:0: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:08.100553:0: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:08.110068:0: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:08.114567:0: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:08.139991:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:08.149506:0: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:08.159016:0: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:08.168530:0: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:08.178037:0: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:08.187551:0: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:08.197061:0: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:08.206575:0: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:08.216082:0: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:08.225596:0: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:08.230097:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:08.235850:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:09.361302:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:09.368596:0: *** Restart logging
MESS:00:00:09.370018:0: brfs: File read: 206 bytes
MESS:00:00:09.379699:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:09.389746:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:09.399776:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:09.409814:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:09.419847:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:09.429885:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:09.439916:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:09.449954:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:09.459987:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:09.470025:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:09.475047:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:09.485670:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:09.495708:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:09.505739:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:09.515777:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:09.525810:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:09.535848:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:09.545879:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:09.555917:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:09.565950:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:09.575988:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:09.581010:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:09.586608:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:09.600400:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:09.610434:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:09.620472:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:09.630502:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:09.640540:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:09.650573:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:09.660611:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:09.670641:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:09.680679:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:09.690712:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:09.695740:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:09.706363:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:09.716396:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:09.726434:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:09.736465:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:09.746503:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:09.756536:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:09.766574:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:09.776604:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:09.786642:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:09.796675:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:09.801703:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:09.807302:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:09.816067:0: HDMI0: hdmi_pixel_encoding: 300000000
MESS:00:00:09.821538:0: HDMI1: hdmi_pixel_encoding: 300000000
MESS:00:00:09.831748:0: dtb_file 'bcm2711-rpi-4-b.dtb'
MESS:00:00:09.838760:0: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
MESS:00:00:09.842008:0: Loading 'bcm2711-rpi-4-b.dtb' to 0x1f0000 size 0xc2a5
MESS:00:00:09.861590:0: brfs: File read: 49829 bytes
MESS:00:00:09.928773:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:09.931337:0: brfs: File read: 206 bytes
MESS:00:00:09.936974:0: brfs: File read: /mfs/sd/overlays/miniuart-bt.dtbo
MESS:00:00:09.962715:0: Loaded overlay 'miniuart-bt'
MESS:00:00:10.020904:0: brfs: File read: 1819 bytes
MESS:00:00:10.022784:0: Failed to open command line file 'cmdline.txt'
MESS:00:00:10.287805:0: brfs: File read: /mfs/sd/RPI_EFI.fd
MESS:00:00:10.290266:0: Loading 'RPI_EFI.fd' to 0x0 size 0x1f0000
MESS:00:00:10.296091:0: No compatible kernel found
MESS:00:00:10.300594:0: Device tree loaded to 0x1f0000 (size 0xc892)
MESS:00:00:10.308245:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:10.315749:0: uart: Baud rate change done...
MESS:00:00:10.317769:0: uart: Baud rate change done...
MESS:00:00:10.330582:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
NOTICE:  BL31: v2.5(release):
NOTICE:  BL31: Built : 18:50:30, Jun 14 2021
UEFI firmware (version UEFI Firmware v1.32 built at 19:05:58 on Oct 19 2021)

ESC (setup), F1 (shell), ENTER (boot)
```

The `ESC (setup), F1 (shell), ENTER (boot)` is shown for at least 10+ minutes and then the UEFI boots. This does not happen if I remove the second USB storage or only keep one connected

jlinton, 2021-12-08:
I'm guessing you're either hitting a power problem, or you have a bad cable/etc that is triggering a known bug in the core edk2 XHCI driver. If it boots fine with either of the devices connected I would find a powered USB hub and see if the problem continues with the devices drawing power from the hub. These RPI's are on the edge when it comes to power consumption and it is easy to get into the territory of unreliable operation simply by drawing too much power from the USB ports.

frezbo, 2021-12-09:
@jlinton yup it was a power problem. I thought the USB were powered from the 5V supply itself. Using a powered hub fixes this. Thanks for pointing me in the right direction. It's a bummer USB port has a limited supply.

closed 2021-12-09 by frezbo